pothosware/PothosSDR #9: Complete GNU Radio + OsmoSDR installer

veso266, 2015-11-16:
Hi Do you think you could add complete GNU Radio + OsmoSDR into Pothos installer?

I would like to use GNU Radio on windows and its a strugle to build everything from source

Thanks for Anwsering and Best Regards

PS: See here: https://pairlist9.pair.net/pipermail/hackrf-dev/2015-November/001632.html

guruofquality, 2015-11-21:
@veso266 What do people generally prefer, gr-qtgui or gr-wxgui? wxgui is just a python dependency, which makes it easy to install. But one of the challenges is that the Qt blocks in GNU Radio use Qt4, whereas Pothos graphics are based around Qt5.

veso266, 2015-11-21:
Hi I personally use gr-wxgui and I think that other people too so

Thanks again for making the installer

guruofquality, 2015-11-23:
New installers are up, here are the docs: https://github.com/pothosware/PothosSDR/wiki/GNURadio

closed 2015-11-23 by guruofquality