pothosware/PothosSDR #10: Bundled GNURadio + GrOsmoSDR

veso266, 2015-11-23:
I have some problems with GNU Radio: 

* when I run it it complains about some theme: C:\Program Files (x86)\gnuradio\bin>python gnuradio-companion.py
gnuradio-companion.py:117: GtkWarning: Could not find the icon 'gnuradio-grc'. T
he 'hicolor' theme
was not found either, perhaps you need to install it.
You can get a copy from:
        http://icon-theme.freedesktop.org/releases
  gtk.window_set_default_icon(gtk.IconTheme().load_icon('gnuradio-grc', 256, 0))

Win32; Microsoft Visual C++ version 12.0; Boost_105900; UHD_003.009.001-0-gf7a15
853

<<< Welcome to GNU Radio Companion 3.7.8.1 >>>

but it works

then when I want to run simple flowgraph to test hackrf TX function I get this error: FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB

and here is how flowgraph looks like: 
[image]

PS: How to install paches like this one: https://github.com/pothosware/PothosSDR/blob/master/patches/volk_cpuid_count_for_msvc.diff and this one: https://github.com/pothosware/PothosSDR/blob/master/patches/gnuradio_fix_filter_truncation.diff ?

Thanks for Anwsering and Best Regards

guruofquality, 2015-11-23:
> gtk.window_set_default_icon(gtk.IconTheme().load_icon('gnuradio-grc', 256, 0))

Its trying to set the corner icon. As far as I know, the icon theme stuff is for linux only. So its always going to print a warning.

> then when I want to run simple flowgraph to test hackrf TX function I get this error: FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB and if I get this to load and TX Light on HackRF is on my program greeze and Python crash

Did you run the zadig.exe instructions (from the tutorial)? Do any of the command line apps for hackrf or soapy sdr work?

> actualy Python crash even if I run something that has gr-wxgui in it

No idea, I had good luck with WX python. I Installed from http://www.lfd.uci.edu/~gohlke/pythonlibs/#wxpython:

* wxPython‑3.0.2.0‑cp27‑none‑win_amd64.whl
* and wxPython_common‑3.0.2.0‑py2‑none‑any.whl

Perhaps its the wx plotters crashing? The screenshot shows working widgets. is that correct?

One guess that I have: Perhaps the plotters are trying to run in OpenGL, mode, which may or may not work. If the following imports work, then it may be trying to use gl:
```
import wx.glcanvas
from OpenGL.GL import *
```

Its just a guess, but if gl present, and is failing to function, it can be disabled. Find gnuradio.conf and set
```
[wxgui]
style=nongl;
```

> PS: How to install paches like this one

All of the patches are built into the installer, it would fail otherwise. I was just specifically mentioning them to bring attention, so maybe someone would help upstream and test important fixes.

veso266, 2015-11-24:
Yes command utilities like hackrf_info and SoapySDRUtil --info work but as I said when I want to run the above flowgraph it is trying to run but when TX light on hackRF shows python crash and I get this error: Python has stopped working and then the program close and I can rerun it from GNU Radio but it cant find my HackRF (I must click reset button on HackRF so that GNU Radio is able to find it)

guruofquality, 2015-11-25:
> Yes command utilities like hackrf_info and SoapySDRUtil --info work but as I said when I want to run the above flowgraph it is trying to run but when TX light on hackRF shows python crash and I get [...]

OK. I understand. I can't reproduce this now. But maybe we can narrow this down. Its good to know if only a particular block has a problem, like osmosdr sink, source, or filter (for example):

1) Does hackrf with osmosdr source work?
2) For the tx problem. To eliminate potential blocks from causing trouble: Are you able to use null source -> "hackrf with osmosdr sink"? Does this simple flow graph also crash?
3) When there is a crash, are you able to get a traceback (using windbg for example). I built all dlls with RelWithDebInfo, so we should get meaningful tracebacks.

> PS: for some reason I can't drag block into a working are I must double click a block to aphear on it

I had the same issue. Actually I had the same issue many years ago with GRC. I remember trying to use a pygtk drag and drop demo application, and it also did not work. I think pygtk windows drag and drop may not be implemented. I know Qt5 DnD works :-)

veso266, 2015-11-25:
Hi sometimes it work or sometimes its just like that: 
[image]

transmitting with null sorce works but when I launch any flowgraph it takes about 10 seconds for hackrf to start transmitting and as soon as anything complex is like this
[image]

GNU Radio crashes 
PS: Did I mention that I can't drag and drop any block not just wxgui one like Waterfall Sink and FFT Sink
PPS: Is this normal but there is no controls in waterfall sink:
[image]

Thanks for Help and Best Regards

guruofquality, 2015-11-26:
Hi, I took a second look at your screenshots, it looks like the wrong driver is being used. I see prints from uhd. Suggested change: set device arguments parameter to ```hackrf=0```. I had the same problem on my setup.

> PS: Did I mention that I can't drag and drop any block not just wxgui one like Waterfall Sink and FFT Sink

Correct. I believe drag and drop from "block tree" to "flow graph" is not possible. The reason is unknown.

> PPS: Is this normal but there is no controls in waterfall sink:

This is my recollection: with pyopengl installed, there is a working waterfall sink. Without pyopengl, the non-accellerated waterfall sink is very bad. Always displays blue. No controls. :-(

If you are successful with installing pyopengl, all the wxgui plotters will look nicer. Please share instructions if successful.

> PS: What did you mean by getting traceback with windbg?

I think this issue is because of driver selection. However, if there is a crash in the future, a traceback from a debugger may be helpful: https://en.wikipedia.org/wiki/WinDbg

veso266, 2015-11-26:
Hi some updates

I got waterfall sink and hackrf to work and suprisingly GNU Radio didn't crash
so for HackRF it was a driver problem when i add hackrf into device arguments HackRF started transmitting immediately no more 10 seconnd wait

for Waterfall sink I just installed OpenGL and it magicly work 
to install OpenGL I downloded these two files: 

http://www.lfd.uci.edu/~gohlke/pythonlibs/chfyvn4n/PyOpenGL_accelerate-3.1.1a1-cp27-none-win_amd64.whl

and 

http://www.lfd.uci.edu/~gohlke/pythonlibs/chfyvn4n/PyOpenGL-3.1.1b1-cp27-none-win_amd64.whl

and then just do from the directory I downloaded file in; pip install PyOpenGL-3.1.1b1-cp27-none-win_amd64.whl PyOpenGL_accelerate-3.1.1a1-cp27-none-win_amd64.whl (if you want to install multiple wheel files you have to use space)

I am still having some problems (there are uuuuu in flowgraph and it say that it can't find any source when I use Audio source: 
[image]

PS: I don't know why it won't work with audio source

Thanks for Help and Best Regards

guruofquality, 2015-11-28:
I have added some new "application notes" here: https://github.com/pothosware/PothosSDR/wiki/GNURadio#application-notes

The install instructions are updated for opengl, including a script for automatic install: https://github.com/pothosware/PothosSDR/wiki/GNURadio#install-dependencies-automatic

> I don't know why it wants to load UHD Driver anyway because isn't this supose to be done when I add UHD Sink/Source not Osmocom

The OsmoSDR blocks also have UHD wrappers. This helps OsmoSDR users to make cross-device applications with the same interface blocks.

> I am still having some problems (there are uuuuu in flowgraph and it say that it can't find any source when I use Audio source: 
PS: I don't know why it won't work with audio source

Can you elaborate? I see the message "INFO: audio source arch: windows"

* Does the audio source produce any samples?
* Or is the audio rate incorrect (not 48k)

Sometimes, even when you have the correct audio rate, you will encounter the 2-clock problem: The audio source 48k is close, but not equal to the hackrfRate/interpolation. If the audio is slower, underflow will happen.

> PPS: Waterfall works and its recieving local FM station but its realy slow?: 

What is slow specifically? The frame/display rate of the plot?

veso266, 2015-11-30:
Hi the audio source doesn't produce any samples (if you mean can I hear anything or not? (if not that: how can I check if there are any samples? (PS: What are Audio source/Sink block device arguments in windows (If you are using jack on linux you just tape in the device that you want to use and then when you start the flowgraph you can connect to it from another program) on windows I would like to use VAC (Virtual Audio cable) and the device that I want to use is "Line 1" in device manager)

on the other note: why does it say (before uuu) No Such file or Directory? (I dont have any wav files here or any directory?

PS: How can I mesure frame rate of the plot in Waterfall Sink?

guruofquality, 2015-12-01:
> Hi the audio source doesn't produce any samples (if you mean can I hear anything or not? (if not that: how can I check if there are any samples? (PS: What are Audio source/Sink block device [...]

I found the problem with the audio source. Its commented out... https://github.com/gnuradio/gnuradio/blob/master/gr-audio/lib/windows/windows_source.cc

However, I think that I can build for Portaudio in the future: https://github.com/gnuradio/gnuradio/blob/master/gr-audio/lib/CMakeLists.txt#L145

> on the other note: why does it say (before uuu) No Such file or Directory?

I don't know which block produces the message. It could be a block that likes to save/load configuration files. Example fftw, volk both open config files.

> PS: How can I mesure frame rate of the plot in Waterfall Sink? 

I'm not sure how to measure, but the rate can be configured by setting "fft rate". This is the number of lines added per second to the plot.

veso266, 2015-12-01:
Hi what did you mean that Audio source problem is commented out (there is a lot lines comented out is it after  // FIXME these should query some kind of user preference (line 54) or after lines 30 and 31?

PS: will there be any difrences in operation if you build GNU Radio with Portaudio?

PPS: How to set fft rate? (hope that this isn's to dummy qestion)

PPS: Do you think you could build with Jack Support?: http://jackaudio.org/faq/jack_on_windows.html

guruofquality, 2015-12-02:
> Hi what did you mean that Audio source problem is commented out (there is a lot lines comented out is it after // FIXME these should query some kind of user preference (line 54) or after lines 30 and 31?

Well technically its not commented out, but the "#if 0" removes 100% of the code. There is no code compiled into the constructor, destructor, or work() function.

> PS: will there be any difrences in operation if you build GNU Radio with Portaudio?

Probably not any differences. Hopefully there will be a working audio source as well.

The way audio selection works in gnuradio is that there is a gr-audio.conf that can be modified to switch between "windows", "portaudio", or any other available support options: https://github.com/gnuradio/gnuradio/blob/master/gr-audio/lib/gr-audio.conf

> PPS: How to set fft rate? (hope that this isn's to dummy qestion)

Set the "fft rate" parameter on the waterfall sink.

> PPS: Do you think you could build with Jack Support?: http://jackaudio.org/faq/jack_on_windows.html

The reason I can build portaudio easily is because portaudio is already used for the Pothos audio source and sink blocks, and so its already available in the installer: https://github.com/pothosware/pothos-audio/wiki

I don't want to add more dependencies if portaudio support works well and solves the problem. Is there any reason for Jack audio vs port audio? Just curious...

veso266, 2015-12-05:
Hi some update

I don't think this is normal but GNU Radio won't play wav files if they are located in the folder that has space in it (when I start the flowgraph I gave me this error:

Generating: "I:\Users\Mitja\SDR\HackRF\GNU Radio Projekti\WBFM\WBFM_Transmitter.py"

Executing: "I:\Users\Mitja\SDR\HackRF\GNU Radio Projekti\WBFM\WBFM_Transmitter.py"

Win32; Microsoft Visual C++ version 12.0; Boost_105900; UHD_003.009.001-0-gf7a15853

Using Volk machine: generic

gr-osmosdr v0.1.4-48-g86ad5842 (0.1.5git) gnuradio 3.7.8.1

built-in sink types: uhd hackrf bladerf soapy 

Using HackRF One with firmware git-44df9d1 

I:\Glasba(s\donna.wav: No such file or directory

UUTraceback (most recent call last):

  File "I:\Users\Mitja\SDR\HackRF\GNU Radio Projekti\WBFM\WBFM_Transmitter.py", line 246, in <module>

    tb = WBFM_Transmitter()

  File "I:\Users\Mitja\SDR\HackRF\GNU Radio Projekti\WBFM\WBFM_Transmitter.py", line 176, in __init__

    self.blocks_file_source_0 = blocks.file_source(gr.sizeof_float*1, "I:\Glasba\50s\donna.wav", True)

  File "C:\Program Files (x86)\gnuradio\lib\site-packages\gnuradio\blocks\blocks_swig0.py", line 523, in make

    return _blocks_swig0.file_source_make(itemsize, filename, repeat)

RuntimeError: can't open file

U

But it will play wav files normaly if there is no space in the directory for instance if there is space in I:\NE ODPIRAJ\bihfolk.wav it won't play if its like that: I:\NEODPIRAJ\bihfolk.wav it will play

I also won't play if there is a "-" a "space" or "01" or some special characters like: "ž" "đ" "š" "ć" "č" 

I don't know why this is happening

PS: Any update on PortAudio support?

Thanks for Anwsering and Best Regards

guruofquality, 2015-12-06:
> But it will play wav files normaly if there is no space in the directory for instance if there is space in I:\NE ODPIRAJ\bihfolk.wav it won't play if its like that: I:\NEODPIRAJ\bihfolk.wav it will play

Its possible that GRC is not generating correctly escaped backspaces. Example ```C:\``` in the parameter box should be written into the python code as ```C:\\```, if not, the backslashes cause a problem. I have two suggestions:

* Test this idea by using forward slash for file paths "/". This should also work.
* If confirmed, please let the GNU Radio developers know about the problem.

> PS: Any update on PortAudio support?

I got it to build portaudio support with some minor patches to use write() API call: https://github.com/pothosware/PothosSDR/blob/master/patches/gnuradio_portaudio_add_io_h.diff

I'm not rebuilding the installers yet. There is some work branches to merge in Pothos first.

veso266, 2015-12-06:
OK Thanks

PS: Is there any way I could apply that patch on working GNU Radio or I would have to rebuild it?

guruofquality, 2015-12-21:
New binaries are available: https://github.com/pothosware/pothos/wiki/Downloads#windows-installer
Announcement: https://groups.google.com/forum/#!topic/pothos-users/skPv3LPCjDI

I have not actually tested for portaudio, so please give it a try. Don't forget about https://github.com/gnuradio/gnuradio/blob/master/gr-audio/lib/gr-audio.conf

If you have trouble, feel free to open a new issue or to mail the list. I am closing this issue for now:
https://github.com/pothosware/pothos/wiki/Support

closed 2015-12-21 by guruofquality