pothosware/PothosSDR #11: Bundled GNURadio + GrOsmoSDR (PortAudio version)

veso266, 2015-12-21:
Hi I've installed pothos again and I've set C:\Program Files (x86)\gnuradio\etc\gnuradio\conf.d\gr-audio.conf and changed from auto to portaudio: [code]
# This file contains system wide configuration data for GNU Radio.
# You may override any setting on a per-user basis by editing
# ~/.gnuradio/config.conf

# specify which audio module to load, or use "auto" to have the system
# select one.  Valid choices depend on your OS and which modules
# you've installed, but typically include:
# auto, alsa, oss, portaudio, jack, osx, windows

[audio]

audio_module = portaudio
[/code]

but it still wan't to use Audio source arch: windows but I know I've setted the right config?
see here: 

Thanks for Anwsering and Best Regards

guruofquality, 2015-12-23:
Ahh... gnuradio constants.cc is loaded full of un-escaped slashes and hard coded paths. I wouldnt count on it finding the conf file at this rate...

constants.cc
```
/* -*- c++ -*- */
/*
 * Copyright 2006,2009,2013 Free Software Foundation, Inc.
 *
 * This file is part of GNU Radio
 *
 * GNU Radio is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 3, or (at your option)
 * any later version.
 *
 * GNU Radio is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with GNU Radio; see the file COPYING.  If not, write to
 * the Free Software Foundation, Inc., 51 Franklin Street,
 * Boston, MA 02110-1301, USA.
 */

#if HAVE_CONFIG_H
#include <config.h>
#endif

#include <gnuradio/constants.h>

namespace gr {

  const std::string
  prefix()
  {
    return "C:\PothosSDR";
  }

  const std::string
  sysconfdir()
  {
    return "C:\PothosSDR\etc";
  }

  const std::string
  prefsdir()
  {
    return "C:\PothosSDR\etc\gnuradio\conf.d";
  }

  const std::string
  build_date()
  {
    return "";
  }

  const std::string
  version()
  {
    return "3.7.8.1";
  }

  // Return individual parts of the version
  const std::string
  major_version()
  {
    return "3";
  }

  const std::string
  api_version()
  {
    return "7";
  }

  const std::string
  minor_version()
  {
    return "8";
  }

  const std::string
  c_compiler()
  {
    return "Microsoft Visual Studio 10.0";
  }

  const std::string
  cxx_compiler()
  {
    return "Microsoft Visual Studio 10.0";
  }

  const std::string
  compiler_flags()
  {
    return "C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe:::/MD /Zi /O2 /Ob1 /D NDEBUG  /DWIN32 /D_WINDOWS /W3 \nC:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe:::/MD /Zi /O2 /Ob1 /D NDEBUG  /DWIN32 /D_WINDOWS /W3 /GR /EHsc";
  }

  const std::string
  build_time_enabled_components()
  {
    return "python-support;gnuradio-runtime;gr-ctrlport;gr-blocks;gnuradio-companion;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-atsc;gr-audio;* portaudio;* windows;gr-channels;gr-noaa;gr-pager;gr-trellis;gr-uhd;gr-utils;gr-vocoder;gr-fcd;gr-wxgui";
  }
} /* namespace gr */
```

```
c:\build\PothosSDR_vc12>c:\PothosSDR\bin\gnuradio-config-info.exe
Program options: c:\PothosSDR\bin\gnuradio-config-info.exe [options]:
  -h [ --help ]         print help message
  --prefix              print GNU Radio installation prefix
  --sysconfdir          print GNU Radio system configuration directory
  --prefsdir            print GNU Radio preferences directory
  --builddate           print GNU Radio build date (RFC2822 format)
  --enabled-components  print GNU Radio build time enabled components
  --cc                  print GNU Radio C compiler version
  --cxx                 print GNU Radio C++ compiler version
  --cflags              print GNU Radio CFLAGS
  -v [ --version ]      print GNU Radio version

c:\PothosSDR\bin\gnuradio-config-info.exe --prefix
C:PothosSDR

c:\PothosSDR\bin\gnuradio-config-info.exe --sysconfdir
C:PothosSDRetc

c:\PothosSDR\bin\gnuradio-config-info.exe --v
3.7.8.1

c:\PothosSDR\bin\gnuradio-config-info.exe --prefsdir
C:PothosSDRetcGrRuntimeconf.d

```

> PS: I have some problems with UDP Source/sink it always trows this error: 

My guess is that WSAInit() is not called. Any socket API will not work.

OK, so more patches.

guruofquality, 2015-12-23:
Actually maybe there is some hope for finding gnuradio.conf...

* It looks like gnuradio will look for the config file in APPDATA/.gnuradio/config.conf
* On my system APPDATA is C:\Users\Me\AppData\Roaming\.gnuradio\config.conf

Although this is still not working for me... references below

app data path
====
```
const char *appdata_path()
  {
    const char *path;

    //first case, try HOME environment variable (unix)
    path = getenv("HOME");
    if(path)
      return path;

    //second case, try APPDATA environment variable (windows)
    path = getenv("APPDATA");
    if(path)
      return path;

    //fall-through case, nothing worked
    return tmp_path();
  }
```

_sys_prefs_filenames
====

```
// Find if there is a ~/.gnuradio/config.conf file and add this to
    // the end of the file list to override any preferences in the
    // installed path config files.
    fs::path homedir = fs::path(gr::appdata_path());
    homedir = homedir/".gnuradio/config.conf";
    if(fs::exists(homedir)) {
      fnames.push_back(homedir.string());
    }
```

guruofquality, 2015-12-23:
> My guess is that WSAStartup() is not called. Any socket API will not work.

Try this: using the import block, ```import socket```. Importing the socket module from python may also call WSAStartup() and cause the UDP blocks to work.

guruofquality, 2015-12-23:
Update, found the issue with the conf file. 

Bug 1) the unescaped slashes in the constants file. They are actually correctly escaped, then the file is overwritten again, without correct escapes.

Bug 2) Because of bug 1), the configuration read code refuses to search any paths, even APPDATA.

The workaround (before recompiling and new installer) would be to hide this python statement somewhere. Possibly edit gnuradio-companion.py and add after the gr import statement. (Modify for your username.)

```
gr.prefs().add_config_file('c:\users\<yourUsername>\AppData\Roaming\.gnuradio\config.conf')
```

veso266, 2015-12-23:
It aphers that GNU Radio registers config because when I delete config.conf it won't run (no error message) 

I edit gnuradio-campanion.py like that: 

[image]

but it still want to use Audio source arch: windows so I don't know how to best check if portaudio is used
regarding UDP Source (if I import socket the error is the same as without it)

Thanks for Anwsering and Best Regards

guruofquality, 2015-12-27:
Work around update
====

Because GRC generates the code and executes in a separate process, the line ```gr.prefs().add_config_file('c:\users\<yourUsername>\AppData\Roaming\.gnuradio\config.conf')``` should be added to this file instead so it becomes part of the generated code ```C:\PothosSDR\lib\site-packages\gnuradio\grc\python\flow_graph.tmpl```. Sorry about the confusion.

Audio update
====

I fixed constants.cc in the gr-runtime project, the fix will be in the next build. For reference see https://github.com/pothosware/gnuradio/issues/22

![Screenshot] (http://i.imgur.com/dC7viBd.png)

C:\Users\User\AppData\Roaming\.gnuradio\config.conf
```
[audio]

audio_module = portaudio
```

Using a audio source to fft sink. port audio worked right away. and its very talkative. Lots of verbose like this, and the verbose from the screenshot:
```
before paHostApiInitializers[0].
after paHostApiInitializers[0].
before paHostApiInitializers[1].
PA - PlatformId = 0x2
PA - MajorVersion = 0x6
PA - MinorVersion = 0x2
PA - PlatformId = 0x2
PA - MajorVersion = 0x6
PA - MinorVersion = 0x2
PA - PlatformId = 0x2
PA - MajorVersion = 0x6
PA - MinorVersion = 0x2
PA - PlatformId = 0x2
```

guruofquality, 2015-12-28:
UDP issue update
====

Ignore my previous comment. This will be fixed in the next build, here is the patch with an explanation of the problem: https://github.com/pothosware/gnuradio/issues/23

guruofquality, 2016-01-10:
Update: I believe this issues are fixed. I working on a build with MSVC 2015, and I will close out this issue when new installers are available. Thanks.

guruofquality, 2016-01-12:
Thanks for all of the feedback and testing. New installers are up with the patches. I am closing out this issue. Please open a new issue if the problem persists, or you find a new issue. Also note that you will still need to set the config.conf to use portaudio. But of course, it will work now :-)

In summary:

* Announcement: https://groups.google.com/forum/#!topic/pothos-users/1HOKTbOPHpo
* The socket issue is tracked here: https://github.com/pothosware/gnuradio/issues/23
* The constants issue is tracked here: https://github.com/pothosware/gnuradio/issues/22

@jmcorgan if you are interested in the issues ^_^

closed 2016-01-12 by guruofquality