pothosware/PothosSDR #14: SDRPlay help

NN4F, 2016-01-12:
Hi, Trying to get the SDRPlay working, i've followed the steps but when I use the soapysdrutil I get

Make device driver=miri
Win32; Microsoft Visual C++ version 12.0; Boost_105900; UHD_003.009.001-0-gf7a15853

Using device #0: Mirics MSi2500 default (e.g. VTX3D card)
usb_open error -12
Error making device: Failed to open mirisdr device.

guruofquality, 2016-01-12:
Since the SDRPlay support is not included in the PothosSDR environment, I'm curious what steps were taken. And can you share what instructions (links) you followed, and what software was downloaded specifically? That will help me to figure this out.

So it looks like you probably got the VC12 installer. I believe that you would have needed to install a mirics dll into the system path, and then a SoapySDRPlay.dll into the SoapySDR modules directory. Its important that the SoapySDRPlay.dll was also built for the same version of MSVC (in this case VC12).

> Using device #0: Mirics MSi2500 default (e.g. VTX3D card)
usb_open error -12
Error making device: Failed to open mirisdr device.

If its possible that everything was installed correctly, and the zadig.exe setup went well. Then I'm not familiar with the print or error coming from the Mirics driver. It may also help to take this one up with the SDRPlay support.

NN4F, 2016-01-12:
Hi Josh,
Thanks for the reply, I followed all the steps on your tutorial page, it sees the device just keeps getting the usb error, I installed zadig, but I've decided not to use the great program, as soon as I install zadig, it makes the sdrplay unusable with all my other software, and this was just a quick dabble and play...if ever it works without zadig, then i'll try again, somehow, charles has cubicsdr running with soapysdr and not having to run zadig....

closed 2016-01-12 by NN4F

guruofquality, 2016-01-12:
To be honest. I don't know if SDR Play needs zadig. It just needs to be recognized by the mirics driver. I just assumed that it was libusb + zadig.

But more importantly, how did you add SDR Play support after installing PothosSDR? That's the part that I am concerned about. Its probably just a mismatch of files.

SDRplay, 2016-01-12:
Now we have the API/EXTIO release done, we are going to be updating the SoapySDRPlay module to behave like the EXTIO plugin. We also plan to make it find the dll on Windows just as the EXTIO plugin does. I should have got to this sooner, but have been tied up with the new gain map and other improvements to the RSP API.

vsboost, 2016-04-06:
Any update to this?

Would like to play around with this.

Thanks