pothosware/PothosSDR #15: Bundled GNURadio + GrOsmoSDR OTA Projects Wish List

veso266, 2016-01-22:
Hi Here I am prepairing a wishlist of some project that you could include into GNURadio portion of Pothos SDR: 

1. GR-RDS: https://github.com/bastibl/gr-rds   
    * The most official and updated one
    * PRIORITY: HIGH
2. GR-DRM: https://github.com/kit-cel/gr-drm
   * most official and updated
   * PRIORITY: LOW
   * PS: I don't know how would you get FAAD2 and FAAC working?? and there are some hier_blocks flowgraphs that you nust open after compiling (linux) so I don't know how would this behave on windows

3. GR-DVBT/S/S2: 
    * GR-DVBT: https://github.com/BogdanDIA/gr-dvbt
    * GR-DVBS: https://github.com/drmpeg/gr-dvbs
    * GR-DVBS2: https://github.com/drmpeg/gr-dvbs2
    * to use theese you would also need gr-qam: https://github.com/argilo/gr-qam
    * PRIORITY: VERY LOW 

PS: What do theese do?: https://github.com/guruofquality/grextras
Thanks and Best Regards

guruofquality, 2016-01-22:
>  GR-DVBT/S/S2: 

I think some of these projects are now included as in-tree here, but I am not sure exactly: https://gnuradio.org/redmine/projects/gnuradio/repository/revisions/master/show/gr-dtv/lib

> PS: What do theese do?: https://github.com/guruofquality/grextras

This was an amazing project that added message passing features using the stream tag architecture. It doesnt compile on 3.7 and most of its functionality is included with gnuradio now.

> PPS: DO you think you could get GNURadio (https://gnuradio.org/redmine/projects/gnuradio/repository/revisions/master/show/gr-zeromq) to work with ZeroMQ: http://zeromq.org/

Probably. There are windows binaries, but since zmq is just another github/cmake project like the rest it should be possible to add it to the build. Looks like I even contributed to the build system in the past/its a small world: https://github.com/zeromq/libzmq/blob/master/CMakeLists.txt

guruofquality, 2016-01-23:
Adding zmq to the build was mostly strait-forward. Had to patch zmq though:

* Commit: https://github.com/pothosware/PothosSDR/commit/21f8b62e88684101217d7a03fd95cbfb8c220bfc
* Upstream issue: https://github.com/zeromq/zeromq4-x/issues/141

guruofquality, 2016-01-25:
Gr-RDS: https://github.com/pothosware/PothosSDR/commit/f4a88f231440d87510f7b963168b1686ddf298e1

guruofquality, 2016-01-25:
@bastibl if you are interested in up-streaming, the patch was very minor: https://github.com/pothosware/PothosSDR/blob/master/patches/gr_rds_msvc_fixes.diff

guruofquality, 2016-01-25:
I know it wasn't on the wishlist, but GQRX is now part of the build!
* commit: https://github.com/pothosware/PothosSDR/commit/d38040667b114f5a1f78f2d0bcda69d1318d12c5
* patch upstreamed: https://github.com/csete/gqrx/commit/fb1d9eebdbc5d07d550e173ce2fea0a4dbec3b47

veso266, 2016-01-25:
@Josh  thanks for all this

Ps: can i use GQRX on windows now?

If I knew it was posible I would add this to wishlist
Best regards

guruofquality, 2016-01-26:
> Ps: can i use GQRX on windows now?

@veso266 Almost, minor issue here: https://github.com/csete/gqrx/issues/332

guruofquality, 2016-01-30:
@veso266 binaries are up for gr 3.7.9, gr-rds, gr-zmq, and gqrx. Give it a shot.

@csete https://github.com/pothosware/PothosSDR/wiki/GQRX

veso266, 2016-01-30:
Hi GQRK works as well as gr-rds
[image]
PS: do you know where is a gain setting?
PPS: If I sellect Configure I/O devices the program crash and I must restart it

Thanks again

csete, 2016-01-30:
@guruofquality Thanks! There is now a link from the gqrx download page to the wiki page.

@veso266 Gain settings are on the "Input controls" tab.

guruofquality, 2016-01-30:
> PPS: If I sellect Configure I/O devices the program crash and I must restart it

I saw that too. But it didnt happen for every device (cant remember which). Perhaps a specific device has a bug in the destructor.

> PS: Do you know if its posible to set some shell associations with GNURadio and .grc

I havent looked into it. Eventually i need to figure this out for the PothosGui .pth files. If you figure it out in the meantime, especially if its just a registry edit, I can add it easily to the installer.

> while GNURadio is a difrent story (when I first try to eyecute any of my flowgraph I get error that xterm is miising and that this message will be shown only once.

I saw that as well. No idea, but since the error went away I ignored it. @skoslowski Any idea?

> If I copy my flowgrphs into folder with one word in it only the flowgraph will run but as soon as I add space and two words into folder the flowgraph will not run and it will show me this error:
C:/Python27/python.exe: can't open file 'C:/test': [Errno 2] No such file or directory

There were some changes to the execute code. I already fixed the path issue: https://github.com/pothosware/gnuradio/issues/27 It looks like the path needs to be quoted as well. So I will probably just change this same patch to add quotes. @skoslowski FYI

guruofquality, 2016-01-31:
@veso266 I updated the grc patch. Since its python, you can manually edit Generator.py to support spaces until I get another build out: https://github.com/pothosware/PothosSDR/blob/master/patches/gnuradio_grc_windows_paths.diff

skoslowski, 2016-01-31:
Thanks, I have already a fix ready (review pending). See https://github.com/skoslowski/gnuradio-wg-grc/commit/73435821ddbb035466589dd4fdcefa0b8785dce7

veso266, 2016-01-31:
Hi I've figured out shell associations with pth files you need to open regedit and go into HKEY_CLASSES_ROOT then create key with .pth name and under defult add whatever name you want I've used PothosSDR.

Then again go into HKEY_CLASSES_ROOT and create another key with the name that u used before (PothosSDR) in there create theese  keys and name them: 
* DefaultIcon (you don't need to do this you just need to do this if you want other icon that is in exe)
* Shell (in there create key with name: Open (in there create key with name: command (in there set Defult value to: "C:\Program Files\PothosSDR\bin\PothosGui.exe" "%1"

or just run theese files that I made: https://github.com/veso266/PothosSDR-Shell-asociasions
(first run AssocPth.reg then run PothosSDR.reg and copy PothosGui.ico into C:\Program Files\PothosSDR\bin\)

PS: if you don't set any icon Icons are kinda small but it works (but if you set icon it will be perfect: 
[image]

Hope it works

guruofquality, 2016-02-01:
@skoslowski Cool. I coped Generator.py wholesale from your branch and it seemed to work out just fine. I will probably have to wait for the next GR release to use it because its a lot of work to patch and test all of this. But its good to know that there is one less patch to worry about :-)

@veso266 I transformed your registry key work into NSIS registry commands. It seems to work. Please enjoy the hilarious amount of escape slashes:

In the CMakeLists.txt
```
set(HLKM_ENV "\\\"SYSTEM\\\\CurrentControlSet\\\\Control\\\\Session Manager\\\\Environment\\\"")

set(Q3 "$\\\\\\\"") #escape quotes within the NSIS command

set(CPACK_NSIS_EXTRA_INSTALL_COMMANDS "
WriteRegStr HKEY_LOCAL_MACHINE ${HLKM_ENV} \\\"POTHOS_ROOT\\\" \\\"$INSTDIR\\\"
WriteRegStr HKEY_LOCAL_MACHINE ${HLKM_ENV} \\\"SOAPY_SDR_ROOT\\\" \\\"$INSTDIR\\\"
WriteRegStr HKEY_LOCAL_MACHINE ${HLKM_ENV} \\\"UHD_PKG_PATH\\\" \\\"$INSTDIR\\\"
WriteRegStr HKEY_LOCAL_MACHINE ${HLKM_ENV} \\\"GRC_BLOCKS_PATH\\\" \\\"$INSTDIR\\\\share\\\\gnuradio\\\\grc\\\\blocks\\\"
WriteRegStr HKEY_CLASSES_ROOT \\\".pth\\\" \\\"\\\" \\\"PothosSDR\\\"
WriteRegStr HKEY_CLASSES_ROOT \\\"PothosSDR\\\\DefaultIcon\\\" \\\"\\\" \\\"$INSTDIR\\\\share\\\\Pothos\\\\icons\\\\PothosGui.ico\\\"
WriteRegStr HKEY_CLASSES_ROOT \\\"PothosSDR\\\\Shell\\\\Open\\\\command\\\" \\\"\\\" \\\"${Q3}$INSTDIR\\\\bin\\\\PothosGui.exe${Q3} ${Q3}%1${Q3} %*\\\"
")
```

To get this in the project.nsi:
```
WriteRegStr HKEY_LOCAL_MACHINE "SYSTEM\CurrentControlSet\Control\Session Manager\Environment" "POTHOS_ROOT" "$INSTDIR"
WriteRegStr HKEY_LOCAL_MACHINE "SYSTEM\CurrentControlSet\Control\Session Manager\Environment" "SOAPY_SDR_ROOT" "$INSTDIR"
WriteRegStr HKEY_LOCAL_MACHINE "SYSTEM\CurrentControlSet\Control\Session Manager\Environment" "UHD_PKG_PATH" "$INSTDIR"
WriteRegStr HKEY_LOCAL_MACHINE "SYSTEM\CurrentControlSet\Control\Session Manager\Environment" "GRC_BLOCKS_PATH" "$INSTDIR\share\gnuradio\grc\blocks"
WriteRegStr HKEY_CLASSES_ROOT ".pth" "" "PothosSDR"
WriteRegStr HKEY_CLASSES_ROOT "PothosSDR\DefaultIcon" "" "$INSTDIR\share\Pothos\icons\PothosGui.ico"
WriteRegStr HKEY_CLASSES_ROOT "PothosSDR\Shell\Open\command" "" "$\"$INSTDIR\bin\PothosGui.exe$\" $\"%1$\" %*"
```

I will try to apply this for GRC as well. Although I wonder if the Open command will work without explicitly specifying the python executable. It should resolve to the python file association. We will find out!

Also I didnt add the start menu shortcut for GRC because CPACK_PACKAGE_EXECUTABLES only supports .exe. So that might need to be handled differently than GQRX/PothosGui were.

guruofquality, 2016-02-03:
So I ended up making a gnuradio-companion.exe with a simple python script and pyinstaller. This creates an executable with an icon thats easy to use with the NSIS start menu and file association command since it was hard to directly call the .py file. See commit: https://github.com/pothosware/PothosSDR/commit/64cbe27c2920bddf0fd4c293a27bda694601f57f

The next build will have working start menu and file associations for GRC and Pothos!

[image]

guruofquality, 2016-05-04:
FYI, I got a second request for gr-drm, so there is a new issue: https://github.com/pothosware/PothosSDR/issues/22

And I made another issue for gr-dvbt request: https://github.com/pothosware/PothosSDR/issues/23

Closing this for now. And installer will be updated for latest GR tag very soon. Thanks

closed 2016-05-04 by guruofquality