pothosware/PothosSDR #17: Module search trys to open files that are not shared libraries

ion-concepts, 2016-02-04:
Noticed that Soapy is pretty indiscriminate (under OSX 10.10) and verbose about trying to open anything under {INSTALL_ROOT}/lib:

```
macbook-cmh:build ianb$ SoapySDRUtil --probe="driver=uhd"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=uhd
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/CMakeMacroLibtoolFile.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/CMakeParseArgumentsCopy.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/FindALSA.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/FindCppUnit.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/FindFFTW3f.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/FindGit.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/FindGSL.cmake"
```

guruofquality, 2016-02-04:
This is more like the windows sdk project. This is the SoapySDR project: https://github.com/pothosware/SoapySDR/issues

So thats odd. What does ```SoapySDRUtil --info``` say? This will tell us where its looking

Its supposed to search
* install_root/lib/SoapySDR/modules
* and $SOAPY_SDR_PLUGIN_PATH

and install_root will be the CMAKE_INSTALL_PREFIX unless this env var is set $SOAPY_SDR_ROOT

ion-concepts, 2016-02-04:
Sorry if you want to move the issue go ahead...I'm schooled for future issues now.

I don't have any $SOAPY prefix enviromentals defined. 
And my install root in CMAKE was : /Users/ianb.
I do have some generic enviromentals such as $LIBRARAY_PATH defined.

```
macbook-cmh:build ianb$ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

API Version: v0.4.1-ge0775c82
ABI Version: v0.4-2
Install root: /Users/ianb
Module found: /Users/ianb/lib/SoapySDR/modules/libuhdSupport.so
Loading modules... Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/CMakeMacroLibtoolFile.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/CMakeParseArgumentsCopy.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/FindALSA.cmake"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/cmake/gnuradio/FindCppUnit.cmake"

----SNIP! repeated ad nauseaum....

Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/uhd/utils/usrp_n2xx_net_burner.py"
Error: EnvironmentError: OSError: dlopen failed to load "/Users/ianb/lib/uhd/utils/usrp_n2xx_net_burner_gui.py"
Mac OS; Clang version 7.0.2 (clang-700.1.81); Boost_105900; UHD_003.010.git-119-g42a3eeb6

done
Available factories...null, uhd, 
```

ion-concepts, 2016-02-04:
FYI. I removed $LIBRARY_PATH and $DYLD_LIBRARY_PATH but no difference

guruofquality, 2016-02-04:
I think SoapySDR is working. It actually shows uhd loaded and doesn't have any recognizable error prints from SoapySDR.

I have a feeling that once SoapySDR loads uhd support that its UHD static initialization trying to load everything in lib/. Those prints are from UHD is what I am saying.

You dont have ```UHD_MODULE_PATH``` set do you?

Just for reference:
* UHD searches here: https://github.com/EttusResearch/uhd/blob/master/host/lib/utils/paths.cpp#L193
* Loads here: https://github.com/EttusResearch/uhd/blob/master/host/lib/utils/load_modules.cpp#L65
* And if you suspect Soapy, it loads here: https://github.com/pothosware/SoapySDR/blob/master/lib/Modules.in.cpp#L184

Unless its a weird install path or env variable set, you might need to add a few prints to see where its looking~

guruofquality, 2016-02-07:
Anyway, I think we figured this out. Just open an issue on the SoapySDR or relevant tracker if it persists. Thanks!

closed 2016-02-07 by guruofquality