pothosware/PothosSDR #38: problem with WX GUI FFT Sink in GNU Radio Companion

pavel-demin, 2017-07-26:
I've just installed PothosSDR-2017.06.21-vc14-x64.exe and python-2.7.13.amd64.msi and I have a problem with some of my GRC flow graphs that contain 'WX GUI FFT Sink'. Even with just 'Noise Source' connected to 'WX GUI FFT Sink' I get the following error message:
```
Executing: C:\Python27\python.exe -u D:\GRC\top_block.py
Traceback (most recent call last):
  File "D:\GRC\top_block.py", line 84, in <module>
    main()
  File "D:\GRC\top_block.py", line 78, in main
    tb = top_block_cls()
  File "D:\GRC\top_block.py", line 58, in __init__
    peak_hold=False,
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\wxgui\fftsink_gl.py", line 92, in __init__
    win=win,
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\fft\logpwrfft.py", line 65, in __init__
    fft = self._fft_block[0](fft_size, True, fft_window)
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\fft\fft_swig.py", line 126, in make
    return _fft_swig.fft_vcc_make(fft_size, forward, window, shift, nthreads)
RuntimeError: The segment is already unlocked.
```

If I replace 'WX GUI FFT Sink' with 'WX GUI Scope Sink', the flow graph works.

guruofquality, 2017-07-26:
Thanks for the extra info. I know that the gr maint branch introduced some lock file for the fftw wisdom file. It looks like its misbehaving :-( I wonder if its causing the problem with gqrx as well which is why I didnt put out another installer this weekend. I might have to revert gr releases and see what happens.

guruofquality, 2017-07-27:
@pavel-demin uploaded a new installer. I think the fft wisdom lock issue was fixed recently in gr maint branch. I also had to fix a problem with the [grc generator](https://github.com/gnuradio/gnuradio/commit/2708f33f). GQRX issue still seems to be present, but I figured it would be good to get out something with some fixes anyway.

guruofquality, 2017-07-31:
GQRX issue turned out to be other bug all together https://github.com/gnuradio/gnuradio/issues/1383 So never mind about that. So both of these were gnuradio bugs either patched out for now or fixed in gr maint branch. New installer uploaded.

closed 2017-07-31 by guruofquality