pothosware/PothosSDR #39: Build for MSVC 2017

guruofquality, 2017-09-10:
* [ ] Possibly blocking dependencies, have to see what Qt has available
  * [ ] Find qt5 development installers for MSVC 2017
  * [ ] Find boost development installers for MSVC 2017
* [ ] Some cmake build strings and that should do it
* [ ] Close when installers for this release are available

Heshyo, 2017-11-28:
Does it mean we can't build using MSVC 2017, and we need MSVC 2015?

Do we need to install the entire Visual Studio, or can we install a specific compiler or something?

guruofquality, 2017-11-28:
Usually this is true across MSVC major releases, they break the ABI. Thats why any major C++ library has to release new prebuilt dlls for each MSVC release (ex boost, qt). I used to build PothosSDR for 2013 and 2015 for this reason. And I am going to build for 2017 as well.

-- But I think in this case MSVC2017 is ABI compatible with 2015, which means we are in luck. I think you can compile under 2017 with 2015 dlls: https://blogs.msdn.microsoft.com/vcblog/2017/03/07/binary-compatibility-and-pain-free-upgrade-why-moving-to-visual-studio-2017-is-almost-too-easy/

>Yes, our binary compatibility guarantee for 2015/2017 applies to all MSVC libraries, including the STL. It’s still best to compile everything consistently with the latest version of VS 2017 (to get all of our correctness and performance fixes), but mixing-and-matching 2015 and 2017 will work. You can even export 2017-only types from a DLL; binaries compiled with 2015 just won’t be able to call those exports (i.e. you’d need to guard your declarations with a version check). -  Stephan T. Lavavej - MSFT

Heshyo, 2017-11-29:
OK thanks. Once I have the time I'll try to compile it with 2017 as I don't have 2015 anymore. I'll report here if you haven't done it first.

guruofquality, 2021-01-06:
2019 branch is in the works, closing 2017

closed 2021-01-06 by guruofquality