pothosware/PothosSDR #40: GNURadio Companion execute failed

noorye4, 2017-09-13:
OS version : windows 10
install version : PothosSDR-2017.08.31-vc14-x64
ErrorMessage:
 Failed to open registry key
HKLM\SOFTWARE\Python\PythonCore\2.7InstallPath
Is Python2.7 installed?

I am counfuse.
I already installed python2.7  and add "C:\python" to my SYSPATH

guruofquality, 2017-09-13:
This might be the result of a python install option. We are using the python registry hooks to know where python2.7 interpreter is. So I have the following suggestion:

From https://github.com/pothosware/PothosSDR/wiki/GNURadio#installation-instructions

> This is a 64-bit installer and GNU Radio requires the 64-bit version of Python 2.7. Look for the "Windows x86-64 MSI installer" on the python downloads page.

> Note: If you choose to customize the python install options, make sure that the "register extension" box is checked. This tells the installer how to find Python at runtime through the windows registry.

So double check those two requirements. Anyway, I hope that explains the issue. We had a big back and fourth discussion on the mailing list and it just turned out that python added this as a variable installation option at some point. Sorry if that caused confusion :-)

noorye4, 2017-09-19:
I solved it.
just install 64bit python
thank you replay.

closed 2019-01-07 by guruofquality