pothosware/PothosSDR #45: Pothos GUI page size is fixed?

wireless-net, 2017-11-29:
Hi,

I am testing out Pothos GUI 0.5.1 PothosSDR-2017.11.26-vc14-x64 on a computer with a 4K display, and while everything appears to display OK, the page size seems very limited -- I only have enough room to place maybe 6 or so blocks. There seems to be an invisible barrier that prevents me from attempting to drag blocks out of this small area.

I cannot find any settings (documented or otherwise) to enlarge this usable area. Any pointers?

By the way, very nice work on this new GUI! Impressive.
--
Regards, Devin

guruofquality, 2017-12-01:
Looks like the available canvas size is hard coded to Full HD size `static const QSizeF GraphDrawCanvasSize(1920, 1080);` I guess the author owns a tiny monitor. I opened an issue to add this: https://github.com/pothosware/PothosFlow/issues/160

guruofquality, 2017-12-31:
A feature to use the screen size and to select it in the graph properties was added and released, its up in the latest build: https://groups.google.com/forum/#!topic/pothos-users/mnhrTo87Vvw

closed 2017-12-31 by guruofquality