pothosware/PothosSDR #46: PATH too long

mycall, 2017-11-30:
I've had success in using this PowerShell script to pack the PATH environment variable, in order to support more directories, and thought I would share it.  

`clear
write-host "Reading..."
write-host

$fso = New-Object -ComObject "Scripting.FileSystemObject"
$shortpaths = @();
$originalPaths = [environment]::GetEnvironmentVariable("path", "Machine").Split(";")

foreach ($path in $originalPaths) {
	$fpath = [System.IO.Path]::GetFullPath("$path");
    $fspath = $fso.GetFolder("$fpath").ShortPath;
	$foundIdx = $shortpaths.IndexOf($fspath);
	if ($foundIdx -gt -1) {	continue; }
	write-host $fpath  -->  $fspath;
	$shortpaths += $fspath;
}
write-host
write-host "Packing..."
write-host
$env:Path = $($shortpaths -join ";");
write-host $env:Path
write-host
write-host "Done." 
write-host`

gvanem, 2017-12-14:
I got the same using this installer: `PothosSDR-2017.11.26-vc14-x64.exe`:

[image]

nlhnt, 2020-04-03:
Any news on that?
I got the same error during the installation.
When I manually added C:\Program Files\PothosSDR\bin to my PATH variable it kind of turned Pothos Flow on, but all the GNURadio blocks I put down are read.

TomCox01, 2020-04-04:
Use Long Path Tool, Long Path Tool can simplify and 

probably end your problems in unlocking, managing and 

renaming files that appear to have a long filename.

einthecorgi2, 2020-08-25:
Same problem

vbursucianu, 2020-11-24:
Got this issue too

n3rdx, 2021-02-27:
yes, I installed the latest I think (PothosSDR-2021.01.28-vc16-x64) today and got the same scary error. I will attempt to re-install it with a slightly shorter (but by how much) pathname. Just thought I would record this on Windows 10. I can install all commercial programs ...

guruofquality, 2021-04-30:
I wonder if this will fix it: https://docs.python.org/3/using/windows.html#removing-the-max-path-limitation

> Windows historically has limited path lengths to 260 characters. This meant that paths longer than this would not resolve and errors would result.
> In the latest versions of Windows, this limitation can be expanded to approximately 32,000 characters. Your administrator will need to activate the “Enable Win32 long paths” group policy, or set LongPathsEnabled to 1 in the registry key HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem.

Skinner927, 2021-10-12:
FWIW, enabling long paths did not change anything.
I have tried installing in `c:\psdr`, but I still get the path error.

It seems to have something to do with adding user or system's `PATH` environment variable rather than file path because if I turn that setting off in the installer, the error goes away.

ratzrattillo, 2022-06-24:
Experienced the same issue. Is this being worked on?
Please inform us, if there is a workaround, or how this is planned to be fixed!

habub, 2022-10-23:
This issue was opened in 2017, now it's 2022, I also have this same issue (2021.07.25-vc16-x64).
Any resolution I missed?

nikstp, 2022-11-01:
I've had no issues installing the software without adding the folder to the path automatically and instead adding it manually; I'd count that as a workaround for now.

Jitteryheart87, 2024-08-02:
how do i add this manually?