pothosware/PothosSDR #47: Wishlist: add gr-nrsc5 to build

veso266, 2017-12-20:
Hi do you think you could add gr-nrsc5: https://github.com/argilo/gr-nrsc5 to GNURadio?

Thanks for Anwsering and Best Regards and Merry Christmas

guruofquality, 2018-04-01:
@veso266 I added this, but I didnt test. Basically it compiled but I have some worries about the internal fdk_aac being built with different flags. It looks like the makefile comes with variables to correct this if that is the case.

@argilo I ended up patching the build slightly. It was missing the PMT library in the set of required components and I had to make some mods for the build calls to the fdk_aac project, diff is here
https://github.com/pothosware/PothosSDR/blob/master/patches/gr_nrsc5_msvc.diff

I was wondering if if would be possible to support optional cmake arguments FDK_AAC_LIBRARY and FDK_AAC_INCLUDE_DIR, when specified would replace the ExternalProject_Add. That way I can build fdk_aac as a separate project with whatever modifications necessary, and gr-nrsc5 would not need a patch for customized build rules. Thoughts?

veso266, 2018-04-01:
Hi I've updated GNURadio now finally (it took a long time for some reason :smile: )

and now I can't use notebooks on WX widgets it complaint:
```python
Param - Grid Position(grid_pos):
	'set' object has no attribute 'append'
```
not sure what the problem is because it worked fine before

guruofquality, 2018-04-01:
> 'set' object has no attribute 'append'
> and now I can't position blocks on WX widgets it complaint:

I just really build the maint branch and hope for the best. It looks like python set type is being used instead of list. List has append, set has add.

@skoslowski Theres no traceback, but any idea about this/recent changes?

> PS: is it possible for GNURadio to use Named pipes in windows somehow?

Probably, from gnuradio's perspective, its just a normal file. Making the named pipe is different.
https://en.wikipedia.org/wiki/Named_pipe#In_Windows

veso266, 2018-04-01:
so how can I fix that set object problem quickly?

guruofquality, 2018-04-01:
@veso266 all of the python is installed, so you can just edit the grc .py files to help debug the problem. Im sure they would be happy to know about the problem and/or get a patch. Aside from that, I just build the project, its a gnuradio bug, you should let them know so the right people are looking at it.

veso266, 2018-04-02:
I fixed it I just copied old PothosSDR/lib/python/site-packages/gnuradio/grc/core/param.py from old PothosSDR installation and now it works

also narrowed the problem down ot this piece of code in param.py
```python
        #########################
        # Grid Position Type
        #########################
        elif t == 'grid_pos':
            if not v:
                # Allow for empty grid pos
                return ''
            e = self.get_parent().get_parent().evaluate(v)
            if not isinstance(e, (list, tuple)) or len(e) != 4 or not all([isinstance(ei, int) for ei in e]):
                raise Exception('A grid position must be a list of 4 integers.')
            row, col, row_span, col_span = e
            # Check row, col
            if row < 0 or col < 0:
                raise Exception('Row and column must be non-negative.')
            # Check row span, col span
            if row_span <= 0 or col_span <= 0:
                raise Exception('Row and column span must be greater than zero.')
            # Get hostage cell parent
            try:
                my_parent = self.get_parent().get_param('notebook').evaluate()
            except:
                my_parent = ''
            # Calculate hostage cells
            for r in range(row_span):
                for c in range(col_span):
                    self.hostage_cells.append((my_parent, (row + r, col + c)))
            # Avoid collisions
            params = filter(lambda p: p is not self, self.get_all_params('grid_pos'))
            for param in params:
                for parent, cell in param._hostage_cells:
                    if (parent, cell) in self.hostage_cells:
                        raise Exception('Another graphical element is using parent "{0}", cell "{1}".'.format(str(parent), str(cell)))
            return e
```
don't know how to fix it and copying old file is probably not a good solution but for now it works and hopefully I didn't break something else along the way :smile:

guruofquality, 2018-04-02:
@veso266 perfect, I was able to git blame and see the issue, the variable was changed from list to type set and I guess it wasnt bug tested: https://github.com/gnuradio/gnuradio/commit/760c147523

As far as letting them know about bugs, I just comment on the problem commit when possible, but you can always open an issue or a pull request. The previous maintainer used to close my pull requests without reading them so I have taken to personally notifying the author of the bug when possible :-)

closed 2018-07-14 by guruofquality