pothosware/PothosSDR #48: bladeRF fails to build with missing pthread.h

eduardosm, 2018-02-05:
bladeRF fails to build, saying that it cannot find `pthread.h`. The file can be found in `C:\PothosSDR-dev\PothosSDR\build\Source\Pthreads`, but the compiler invocation does not seem to include `/I"C:\PothosSDR-dev\PothosSDR\build\Source\Pthreads"`.

```
 C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64\CL.exe /c /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\include" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\l
  ibbladeRF\src" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\..\fpga_common\include" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\..\firmware_common" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\common\
  include" /I"C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\common\include" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\common\include\windows" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_te
  st\test_config_file\..\common\include" /Zi /nologo /W3 /WX /O2 /Ob1 /D WIN32 /D _WINDOWS /D NDEBUG /D LOG_INCLUDE_FILE_INFO /D _CRT_SECURE_NO_WARNINGS /D "LIBBLADERF_SEARCH_PREFIX=\"C:/PothosSDR\"" /D LOGGING_ENABLED=1 /D "CMAKE_INTDI
  R=\"RelWithDebInfo\"" /D _MBCS /Gm- /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"libbladeRF_test_config_file.dir\RelWithDebInfo\\" /Fd"libbladeRF_test_config_file.dir\RelWithDebInfo\vc140.pdb" /Gd /TC /errorReport:queue
   "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\common\src\test_common.c" "C:\PothosSDR-dev\
  PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\bladerf_priv.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\config.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libb
  ladeRF\src\dc_cal_table.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\file_ops.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\flash.c" "C:\PothosSDR-dev\PothosSDR\
  build\Source\bladeRF\host\libraries\libbladeRF\src\flash_fields.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\fpga.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\s
  i5338.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\tuning.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\version_compat.c" "C:\PothosSDR-dev\PothosSDR\build\Sourc
  e\bladeRF\host\libraries\libbladeRF\src\xb.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\common\src\conversions.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\common\src\log.c" "C:\PothosSDR-dev\PothosSDR\build\So
  urce\bladeRF\host\common\src\parse.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\fpga_common\src\band_select.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\fpga_common\src\lms.c"
    main.c
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    test_common.c
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\common\include\windows\clock_gettime.h(28): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librari
es\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    bladerf_priv.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    config.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    dc_cal_table.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    file_ops.c
    flash.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    flash_fields.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    fpga.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    si5338.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    tuning.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    version_compat.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    xb.c
c:\pothossdr-dev\pothossdr\build\source\bladerf\host\libraries\libbladerf\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    conversions.c
    log.c
    parse.c
    band_select.c
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
    lms.c
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\bladerf_priv.h(35): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\librar
ies\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
```

guruofquality, 2018-02-05:
Bug confirmed. I actually just ran into this last night when I was attempting to build a new installer. It might have been caused by a bladerf update (some new bug in the cmakelists for bladerf), but I haven't narrowed down yet...

eduardosm, 2018-02-05:
As a workaround, I copied pthread headers to bladerf's include directory. Now I'm getting this:
```
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64\CL.exe /c /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\include" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\l
  ibbladeRF\src" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\..\fpga_common\include" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\..\firmware_common" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\common\
  include" /I"C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\common\include" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\common\include\windows" /I"C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_te
  st\test_config_file\..\common\include" /Zi /nologo /W3 /WX /O2 /Ob1 /D WIN32 /D _WINDOWS /D NDEBUG /D LOG_INCLUDE_FILE_INFO /D _CRT_SECURE_NO_WARNINGS /D "LIBBLADERF_SEARCH_PREFIX=\"C:/PothosSDR\"" /D LOGGING_ENABLED=1 /D "CMAKE_INTDI
  R=\"RelWithDebInfo\"" /D _MBCS /Gm- /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"libbladeRF_test_config_file.dir\RelWithDebInfo\\" /Fd"libbladeRF_test_config_file.dir\RelWithDebInfo\vc140.pdb" /Gd /TC /errorReport:queue
   "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\common\src\test_common.c" "C:\PothosSDR-dev\
  PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\bladerf_priv.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\config.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libb
  ladeRF\src\dc_cal_table.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\flash.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\flash_fields.c" "C:\PothosSDR-dev\Pothos
  SDR\build\Source\bladeRF\host\libraries\libbladeRF\src\fpga.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\si5338.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\tun
  ing.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\version_compat.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF\src\xb.c" "C:\PothosSDR-dev\PothosSDR\build\Source\blad
  eRF\fpga_common\src\band_select.c" "C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\fpga_common\src\lms.c"
    main.c
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c(67): error C2220: warning treated as error - no 'object' file generated [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\libraries
\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c(67): warning C4244: 'initializing': conversion from 'double' to 'unsigned int', possible loss of data [C:\PothosSDR-dev\PothosSDR
\build\Build\bladeRF\libraries\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c(78): warning C4244: 'initializing': conversion from 'double' to 'unsigned int', possible loss of data [C:\PothosSDR-dev\PothosSDR
\build\Build\bladeRF\libraries\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c(89): warning C4244: 'initializing': conversion from 'double' to 'unsigned int', possible loss of data [C:\PothosSDR-dev\PothosSDR
\build\Build\bladeRF\libraries\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c(215): warning C4244: '=': conversion from 'uint64_t' to 'unsigned int', possible loss of data [C:\PothosSDR-dev\PothosSDR\build\B
uild\bladeRF\libraries\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c(291): error C2057: expected constant expression [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\libraries\libbladeRF_test\test_co
nfig_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c(291): error C2466: cannot allocate an array of constant size 0 [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\libraries\libblade
RF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
C:\PothosSDR-dev\PothosSDR\build\Source\bladeRF\host\libraries\libbladeRF_test\test_config_file\src\main.c(291): error C2133: 'filename': unknown size [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\libraries\libbladeRF_test\test_config
[... cut ...]
k. See online help for details. [C:\PothosSDR-dev\PothosSDR\build\Build\bladeRF\libraries\libbladeRF_test\test_config_file\libbladeRF_test_config_file.vcxproj] [C:\PothosSDR-dev\PothosSDR\build\bladeRF.vcxproj]
```
It seems that `bladeRF\host\libraries\libbladeRF_test\test_config_file` has been added recently [1], so I will try to disable that part (commenting `add_subdirectory(test_config_file)` in `CMakeLists.txt`).

[1] https://github.com/Nuand/bladeRF/commits/master/host/libraries/libbladeRF_test/test_config_file

eduardosm, 2018-02-05:
It seems it worked, now its building GrOsmoSdr and GQRX. I believe that, if everything builds fine, the easiest solution is just to disable `test_config_file`.

guruofquality, 2018-02-06:
I just disabled the test config file: https://github.com/pothosware/PothosSDR/commit/fc250da7b88d8d2e8c8e65a8bdd2718408489f17

closed 2018-02-06 by guruofquality