pothosware/PothosSDR #50: Issue with 2018.03.12 release

KarlL2, 2018-03-22:
I installed my LimeSDR PCIe on a new PC and installed the latest build _PothosSDR-2018.03.12-vc14-x64.exe_ . In GQRX I can listen at 40MS/s, but anything higher gives me tons of noise, the spectrum becomes all yellow.

In my own C++ program using SoapySDR, I often have errors while receiving on one channel (`readStream` returns -1), and when I try to use 2 channels, I get usually 3 warnings `Channel alignment failed` and then I can't receive anything.

After updating the drivers, BIOS and all, I uninstalled Pothos and installed the previous build _PothosSDR-2018.02.05-vc14-x64.exe_ , like on the previous PC, and everything went back to working normally.

guruofquality, 2018-03-22:
Keep in mind that I'm just building the master branch of LimeSuite. If there's some kind of regression in LimeSuite for this hardware, the developers need to know about it https://github.com/myriadrf/LimeSuite . I can always rebuild to get new fixes of course.

KarlL2, 2018-03-22:
Thanks Josh, it makes sense. I'll add something in LimeSuite instead.

closed 2018-03-22 by KarlL2