pothosware/PothosSDR #51: Does not install in different directory

zanco, 2018-04-21:
Hi. Windows 10 64 bit trying to install.
Get an error "can not write to file" C:\Program Files\PothosSDR\bin\SoapySDR.dll"

Can ignore, after which more files fail.
Did the same installation as administrator with the same error(s).

Tried to install in different folder, C:\benno\pothossdr but then installation completes but starting the software gives errors about missing files in C:\Program Files\PothosSDR but that is right because that is not where I installed it.

Any help?

Thanks,
Ben

guruofquality, 2018-04-21:
Sounds like something on the system like another application was reading/using the file at the time. All I can recommend is rebooting, manually deleting anything left from the previous install and retrying. Its a typical NSIS generated installer, it just tries to copy files and bombs out if it cant for some reason.

zanco, 2018-04-21:
After a reboot it installed without issues.

Thanks

closed 2018-04-21 by zanco