pothosware/PothosSDR #53: SoapySdR please include drivers for plutosdr --windows 10....

guruofquality, 2018-04-28:
_From @loid123 on April 28, 2018 0:38_

Please include drivers for libiio, in the windows binary of pothos, so that adalm-pluto [plutosdr] can be used in windows with pothos...

Thanks.

_Copied from original issue: pothosware/PothosCore#159_

beta-tester, 2018-07-01:
yes, i have the same desire...
please add PlutoSDR !!!

it would give me the ability to use my PlutoSDR in the [welle.io project](https://www.welle.io/) under Windows.

guruofquality, 2018-07-02:
The IIO zip doesnt look too bad, this might go smoothly

* add cmake code to download, unzip, install: windows zip for iio binaries - https://github.com/analogdevicesinc/libiio/releases
* Add PothosIIO - https://github.com/pothosware/PothosIIO/wiki
* Add PlutoSDR - https://github.com/jocover/SoapyPlutoSDR

beta-tester, 2018-07-03:
under Linux it is no problem to me to compile things,
but under windows i have no plan how to get it work because of no ability/permission on that computer to install all the dev dependencies/environment to compile necessary stuff.
so i need compiled binaries.

i have [PlutoSDR-M2k-USB-Drivers.exe](https://github.com/analogdevicesinc/plutosdr-m2k-drivers-win/releases/download/v0.7/PlutoSDR-M2k-USB-Drivers.exe) installed for windows already,
i have downloaded and installed [PothosSDR-2018.04.26-vc14-x64](http://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2018.04.26-vc14-x64.exe)

i can use [SDRangel](https://github.com/f4exb/sdrangel/releases) with my PlutoSDR without problems.
i think they have native support for PlutoSDR.

guruofquality, 2018-07-03:
Took a while but building so far: https://github.com/pothosware/PothosSDR/tree/plutosdr

* already had libusb and Pthreads
* libxml2, libiio, and libad9361 added
* it builds and doesnt error out when loading the pluto module

Have to see if I can get something out for testing

beta-tester, 2018-07-05:
thank you.

now i can see "plutosdr #0 192.168.2.1" in the "SoapySDR Source".

but unfortunately i don't get any output...
there is an error message

**Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.6\proxy\environment\PythonSupport.dll): failed safe load**

i am not sure if this is the reason.
i had installed version "PothosSDR-2018.04.26-vc14-x64.exe" before, and as it didn't aupported my PlutoSDR i uninstalled it. (but i think to remember, that is worked with an RTL donlge at that time)
now i installed the version "PothosSDR-2018.07.04-vc14-x64.exe" that shows the PlutoSDR, but with that error message. even when i use the RTL dongle again i get that error.
also trying to uninstall and reinstall the older version i now always get that error...

now i get none of them to work under Pothos Flow...

did i missed something?

a simple rf test with PlutoSDR:
[image]
[image]

and with RTL Dongle:
[image]

loid123, 2018-07-05:
When I try to install on my machine, Windows Defender keeps popping up warning me not to install.
This did not, and does not happen when I try to install the 3 earlier versions.

[p1]<https://user-images.githubusercontent.com/4750719/42316593-87656256-804a-11e8-8905-0c2ffa0c0f47.PNG>
[p2]<https://user-images.githubusercontent.com/4750719/42316603-8db5d2bc-804a-11e8-96a8-459354d6bf26.PNG>

[p3]<https://user-images.githubusercontent.com/4750719/42316609-91e43982-804a-11e8-9d33-fe998b7ec156.PNG>

loid123, 2018-07-05:
It is stock windows defender.
My virus scanner found nothing and so far is allowing me to run, without deleting anything so far.
Hopefully its just certificate signing issue.

GQRX recognizes plutosdr automatically and works fine with sample rate of 3 Mhz and bandwidth of 3Mhz.

Will test it further and also cubicsdr.

Pothos gave me this:

“Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.6\proxy\environment\PythonSupport.dll): failed safe load “

I checked the box with Python version 2 when installing.

When I loaded soapysdr it found plutosdr and loaded usb settings automatically.

Thank you all so much…

@beta-tester<https://github.com/beta-tester> do you what boxes you checked in the installer. Its nothing to worry about, it usually means that python dlls arent present. I just want to double check that you checked these boxes, otherwise, I need to fix the install rules so they areint installed by default.

I don't have any experience with pluto, so if it connects to the device that seems like a very good sign that the drivers work. Maybe there are some recommended gain settings? You might try CubicSDR/GQRX too.

@loid123<https://github.com/loid123> I have no idea. I using windows 10 too with defender on and its not saying anything. But that always a concern, so I will try a few other machines. Is it like a particular virus scan speaking through windows defender, or just the stock windows defender?

loid123, 2018-07-05:
Plutosdr not recognized by cubicsdr.
Then again neither is it in the official version.

guruofquality, 2018-07-05:
> Plutosdr not recognized by cubicsdr. Then again neither is it in the official version.

CubicSDR is really agnostic to the device, so if its detected by SoapySDRUtil, it should work. The official version could easily just not be packaging the hardware support. No idea...

> I checked the box with Python version 2 when installing.

Its actually a development thing for writing python blocks for Pothos, its not the same as general python2 bindings for lets say gnuradio or soapysdr. So you don't need to check this.

loid123, 2018-07-05:
Got the pothos example simple_fm_demod.pothos working with plutosdr.

Thank you.  Thank you!

beta-tester, 2018-07-05:
> @beta-tester do you what boxes you checked in the installer. Its nothing to worry about, it usually means that python dlls arent present. I just want to double check that you checked these boxes, [...]

no, i didn't ticked the checkboxes for python... i thought i hade already python installed before.
now i reinstalled Pothos with ticking Python2 and Python3 support.

ok, now i get output from PlutoSDR. curiously not with the first run. i changed some settings (gain: auto, bandwidth to 2.5e6) and reload framework. after that it was working.
maybe i did something wrong previously.

but now i get a waterfall:
[image]

[image]

PS.: now i get tree of those error messages... maybe one for each python version ;-)
PPS.: the bundeled CubicSDR and Gqrx works with PlutoSDR now. 
[image]
(in Gqrx i only can tune down to 140MHz but maybe i missed some setting - is my first use of Gqrx)

now i have to see, how i get SoapySDR to work in the [welle.io project](https://www.welle.io/)...
i hoped with simply installing PothosSDR it would be done.

but anyhow, thank you very much for your work!

loid123, 2018-07-05:
I ran SoapySDRUtil –find, it found plutosdr.

Then I started Cubicsdr again and plutosdr showed up in its list of sdr devices.

Cubicsdr works fine with plutosdr.

Thanks again.

[p1b]<https://user-images.githubusercontent.com/4750719/42337978-a2a5496c-8088-11e8-8130-fa97482bde55.PNG>

[p2b]<https://user-images.githubusercontent.com/4750719/42337984-a5f0f4c2-8088-11e8-81b6-f62a7b8c83f1.PNG>

[p3b]<https://user-images.githubusercontent.com/4750719/42338530-67cdf350-808a-11e8-82ea-e5948d061f0f.PNG>

beta-tester, 2018-07-05:
> When I try to install on my machine, Windows Defender keeps popping up warning me not to install.

yes, i also got that warning from windows defender. i made a right-click to the EXE file opened the properties and ticked the checkbox to allow access to that file.

[image]

guruofquality, 2018-07-05:
> no, i didn't ticked the checkboxes for python... i thought i hade already python installed before.
now i reinstalled Pothos with ticking Python2 and Python3 support.

@beta-tester Just a few more possibilities in case you have trouble when you see this message
* the PythonSupport dlls could stick around in an old install. The installer does not uninstall if an existing folder is present, so sometimes old files cause complications.
* the python runtime (python.dll) dlls also need to be in the PATH which they may not be by default. So you will see the error if the PythonSupport.dll is installed, but no python.dll in the %PATH%

You can always delete the file in question. That will fix the print :-)

> yes, i also got that warning from windows defender. i made a right-click to the EXE file opened the properties and ticked the checkbox to allow access to that file.

I tried on a few machines since then, and I'm not seeing that defender issue. But its good that the unpacked install is not causing any issues with defender.

For some reason the installer must look suspicious because has so many 100s of dlls from different projects. Or the issue just "installer signing", which is not something I have done. So maybe we have different security settings for defender.

guruofquality, 2018-07-19:
Can anyone comment on https://github.com/jocover/SoapyPlutoSDR/issues/3 ? There seems to be some huge timeout issue when pluto driver access hardware thats not present.

Is it a network device or a usb device? Looks like USB.

* If its a usb device, that hole section of network stuff can be commented out
* if it is a network device, it needs to only do that when an actual host is specified.

beta-tester, 2018-07-20:
the PlutoSDR is connected to the computer via USB port,
but it will apper to the compuer as a USB-networkinterface (and USB-serial port and USB-memory-stick).
by default at IP 192.168.2.1 and hostname "pluto", but hostname and IP address can be changed on the PlutoSDR.

there is also an option to put an USB-WLAN dongle to the PlutoSDR and let the PlutoSDR completely run without a direct physical electrical conection to a computer. then it is connected to the computer by WiFi only.

so the communication to the pluto is done mainly via network connection.