pothosware/PothosSDR #55: Updating SoapySDR to 0.7

jlobingier, 2018-05-22:
I built my Windows system with "PothosSDR-2018.04.26-vc14-x64.exe " which installed support for the 0.6 Lib version.  How do I update this install to support 0.7?

Thank you,

Jack

guruofquality, 2018-05-22:
You can manually build SoapySDR master branch + whatever modules you need. And install over the c:\program files\PothosSDR directory (use the same CMAKE_INSTALL_PREFIX)

Or if you are building PothosSDR, you can change this line to master: https://github.com/pothosware/PothosSDR/blob/master/BuildSoapySDR.cmake#L22

PS The 0.7 stuff is the currently unreleased master branch? Is there a particular feature you are looking that missing in the 0.6 API?

guruofquality, 2018-11-05:
Version with soapy0.7 up

closed 2018-11-05 by guruofquality