pothosware/PothosSDR #62: New release of Soapy/Pothos for BladeRF v2?

OhSoGood, 2018-11-02:
Hi 
Nuand has released a new version of its libraries for its BladeRF v2 (see: https://www.nuand.com/win_installers/ ). Could you publish an updated version of Soapy and Pothos which handle this new version, even if it's just a daily non-official build?
Thank you!

guruofquality, 2018-11-03:
Does this mean that the Bladerfv2 not working in the most recent installer?

OhSoGood, 2018-11-03:
No, it doesn't. Furthermore, Nuand team has added some new functionnality

OhSoGood, 2018-11-03:
No, it doesn't. Furthermore, Nuand team has added one new functionality that I would really to get via Soapy as I need it for one project (in their commits of 21/10/2018 https://github.com/Nuand/bladeRF/commits/master ).

I pulled and compiled SoapyBladeRF, with the latest Pothos (2018-09-09) and bladerf v2 (bladeRF-win-installer-2018.10-rc1.exe) and their includes, but Pothos refused to load it.

If you could publish a new version ofthe whole Pothos package, that would be really welcomed!

guruofquality, 2018-11-05:
I uploaded a new installer, it builds bladerf and soapy support from master

OhSoGood, 2018-11-05:
Triple thanks.
Just a little quirk, the bandwidths are hardcoded in bladeRF_SoapySDR::listBandwidths at https://github.com/pothosware/SoapyBladeRF/blob/master/bladeRF_Settings.cpp#L748 and corresponds to the blade v1. But that may be informative only and without real impact. I don't know which values should be used for the v2.

OhSoGood, 2018-11-05:
Similarly, SoapyBladeRF uses constants defined in bladeRF1.h which corresponds to the device version 1. Ex:
BLADERF_BANDWIDTH_MAX is 28MHz but should be 56MHz. 
BLADERF_FREQUENCY_MAX is 3,8GHz but should be 6GHz.

guruofquality, 2018-11-05:
Can you make this an issue on the SoapyBladeRF project (or PR if you know how to fix it) :-)

I guess closing this since the installer is updated, thanks!

closed 2018-11-05 by guruofquality

OhSoGood, 2018-11-05:
Issue opened at [#22](https://github.com/pothosware/SoapyBladeRF/issues/22)