pothosware/PothosSDR #63: GnuRadio bundled problem

pietro62, 2019-02-05:


guruofquality, 2019-02-05:
GQRX is also using the osmosdr source block. So if GQRX works, this is very unexpected. Can you double check that the arguments used to configure the device in GQRX are the same? device args, sample rate, etc?

pietro62, 2019-02-06:
Hi guruofquality, thank you for fast answer .... well using version PothosSDR-2019.01.26-vc14-x64.
I've double - triple checked but nothing ! Is SoapySDRPlay bundled in GnuRadio ? I have only osmosdr block.  It's possible to check if osmocom module (soapy=1,driver=sdrplay) is streaming out some data ? 
Best regards ... Pietro

[image]

[image]

[image]

[image]

pietro62, 2019-02-06:
No errors ..... but nothing showing on FFT (or other sink ....)

[image]

pietro62, 2019-02-06:
[image]

Just to confirm that all the rest is working OK (read a wav file at 5 Msps recorded with HDSDR).

guruofquality, 2019-02-09:
It is streaming, you are just looking at zeros probably. Change your BB gain to under 10. This worked for me. Not sure why, have to ask the manufacturer. Maybe a safety squelch? Digital multiplier gone wrong?

closed 2019-02-09 by guruofquality

pietro62, 2019-02-09:
Wooowww !! I confirm it's just working great ! Now need just to start building some test flows ! Btw, what's the meaning of BB Gain ? thank you so much. Regards Pietro