pothosware/PothosSDR #64: PothosSDR osmocom, another bug ?

pietro62, 2019-02-09:
Hello,

after being able to run SDRPlay hardware with osmocom source (soapy=4,driver=sdrplay) on bundled GnuRadio (installer version 2019.01.26) see me other post (thank you very much for your effort, bb gain  must be < 10), probably I found another bug with this configuration:  with any sample rate ( 2, 2.048, 5 msps etc) on a simple RX with osmocom and FFT sink, I got only the center +/- 200 kHz spectrum, nothing else (see picture 1). I try any possible config on osmocom block but nothing help. Very frustrating. The strange thing is that after being using GRC with this behavior, closing GRC and starting GQRX, I have the same problem and situation: small bandwidth !!! Only solution is to change GQRX to other input rate (see pictures 2 and 3). Any help please ? 

[image]
[image]
[image]

pietro62, 2019-02-09:
Update: the very same happens on fresh W7-x64 install starting GQRX. Every start il like image 2, changing Input Rate on control panel, solve the problem as you might see on image 3 .... mumble mumble ...

pietro62, 2019-02-22:
Update: I think, when installing & using GQRX or GNUradio with Osmosdr module, the default value for BandWidth is set to 0.0000 MHz. As result I had a small RF bandwith as in picture nr. 1 and 2. So it's necessary to set this value to 1.536 MHz or more to achieve good results, as depicted here below.

[image]

guruofquality, 2019-03-09:
There is nothing I can do about defaults in these programs or how the software bounds checks. You will have to take this up with the specific issue tracker or make a pull request with the fix.

closed 2019-03-09 by guruofquality