pothosware/PothosSDR #67: Invalid cert when accessing  https://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D

vsboost, 2019-03-16:
https://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D Unable to communicate securely with peer: requested domain name does not match the server’s certificate.

guruofquality, 2019-03-16:
@9600 FYI You have bad https certs on the myriadrf downloads server. Probably not a big deal. https://letsencrypt.org/ has been great for getting free ssl certs. Thanks.

@vsboost I didnt even know it was running https :-P

9600, 2019-03-21:
@guruofquality It's not. The issue is that the new myriadrf.org website has strict HTTPS policy and so once you have visited this, your browser expects HTTPS on sub-domains. However, the only SSL vhost on that IP is for the wiki and hence the error when your browser redirects. It's a misleading error...

Looking into the implications of moving downloads to HTTPS only. There is a staging server at https://downloads.stage.myriadrf.org and @ztamosevicius is testing LimeSuiteGUI and LimeUtil with it.

In the meantime you can use a private/incognito window to access the existing HTTP host.

guruofquality, 2019-03-25:
@9600 The staging server looks good. I didnt see a way to provide a URL that would sort by newest first. I found that since I started providing the link with `?C=M;O=D`, that I stopped getting support issues from people installing the absolute oldest exe :-P Anyway, if you had the option, I would make it sort by either newest by default or alphabetically sorted backwards just for this reason.

9600, 2019-03-25:
@guruofquality updated to sort on date with most recent at the top.

closed 2019-04-04 by vsboost