pothosware/PothosSDR #69: HackRF issues

keyboarderror, 2019-06-06:
Experiencing issues related to using the HackRF. I can't seem to access hardware properly in GNURadio though other things like audio sources and file sinks work. Other programs that use PothosSDR components in Windows such as SDRAngel are also experiencing problems. See https://github.com/f4exb/sdrangel/issues/353 It should be noted receive works there. All works normally in Linux. Hoping for additional input.

guruofquality, 2019-06-08:
Do the hackrf guys know that there is a windows issue? Maybe someone accidentally broke it with some portability issue. There isnt much I can do other than build libhackrf. I don't have my hackrf on me now, so it will be a few days before I can try this out.

How does the command line streaming fair?
```
SoapySDRUtil --rate=1e6 --direction=TX --args="driver=hackrf"
```

keyboarderror, 2019-06-09:
I have no idea. I've been tracing the problem over the last few months. 
Here's the output:

SoapySDRUtil --rate=1e6 --direction=TX --args="driver=hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Opening HackRF One #0 325866e621553723...
Stream format: CS8
Num channels: 1
Element size: 2 bytes
Begin TX rate test at 1 Msps
Starting stream loop, press Ctrl+C to exit...
1.38621 Msps    2.77241 MBps
1.19297 Msps    2.38594 MBps
1.12861 Msps    2.25722 MBps
1.09644 Msps    2.19288 MBps
1.07714 Msps    2.15427 MBps
1.06427 Msps    2.12854 MBps
1.05508 Msps    2.11016 MBps
1.04819 Msps    2.09638 MBps
1.04283 Msps    2.08566 MBps
1.03854 Msps    2.07709 MBps
1.03504 Msps    2.07007 MBps
1.03211 Msps    2.06423 MBps
1.02964 Msps    2.05928 MBps
1.02752 Msps    2.05504 MBps
1.02568 Msps    2.05136 MBps
1.02407 Msps    2.04815 MBps
1.02266 Msps    2.04531 MBps
1.02139 Msps    2.04279 MBps
1.02027 Msps    2.04053 MBps
1.01925 Msps    2.0385 MBps
1.01833 Msps    2.03666 MBps
1.0175 Msps     2.03499 MBps
1.01673 Msps    2.03347 MBps
1.01604 Msps    2.03207 MBps
1.01539 Msps    2.03079 MBps
1.0148 Msps     2.0296 MBps
1.01425 Msps    2.0285 MBps
1.01374 Msps    2.02748 MBps
1.01326 Msps    2.02653 MBps
1.01282 Msps    2.02564 MBps
1.01241 Msps    2.02481 MBps
1.01202 Msps    2.02403 MBps
1.01165 Msps    2.0233 MBps
1.01131 Msps    2.02262 MBps
1.01098 Msps    2.02197 MBps
1.01068 Msps    2.02135 MBps
1.01039 Msps    2.02078 MBps
1.01011 Msps    2.02023 MBps
1.00985 Msps    2.01971 MBps
1.00961 Msps    2.01921 MBps
1.00937 Msps    2.01874 MBps
1.00915 Msps    2.01829 MBps
1.00893 Msps    2.01787 MBps
1.00873 Msps    2.01746 MBps
1.00853 Msps    2.01707 MBps
1.00835 Msps    2.0167 MBps
1.00817 Msps    2.01634 MBps
1.008 Msps      2.016 MBps
1.00783 Msps    2.01567 MBps
1.00768 Msps    2.01535 MBps
1.00753 Msps    2.01505 MBps
1.00738 Msps    2.01476 MBps
1.00724 Msps    2.01448 MBps
1.00711 Msps    2.01421 MBps
1.00698 Msps    2.01395 MBps
1.00685 Msps    2.0137 MBps
1.00673 Msps    2.01346 MBps
1.00661 Msps    2.01322 MBps
1.0065 Msps     2.013 MBps
1.00639 Msps    2.01278 MBps
1.00629 Msps    2.01257 MBps
1.00618 Msps    2.01237 MBps
1.00608 Msps    2.01217 MBps
1.00599 Msps    2.01198 MBps
1.0059 Msps     2.01179 MBps
1.00581 Msps    2.01161 MBps
1.00572 Msps    2.01144 MBps
1.00563 Msps    2.01127 MBps
1.00555 Msps    2.0111 MBps
1.00547 Msps    2.01094 MBps
1.00539 Msps    2.01079 MBps
1.00532 Msps    2.01064 MBps
1.00525 Msps    2.01049 MBps
1.00517 Msps    2.01035 MBps
1.0051 Msps     2.01021 MBps
1.00504 Msps    2.01007 MBps
1.00497 Msps    2.00994 MBps
1.00491 Msps    2.00981 MBps
Etc...

Controller is a Fresco Logic USB 3.0 EHC I originally got for FL2k and stuff.

keyboarderror, 2019-06-13:
To help isolate specific issues, this is the log I get for GRC in Windows. The paths and files appear to be correct.

Generating: 'C:\\SDR\\NBFM_Transmitter.py'

Executing: C:\Program Files\GNURadio-3.7\gr-python27\python.exe -u C:\SDR\NBFM_Transmitter.py

Traceback (most recent call last):
  File "C:\SDR\NBFM_Transmitter.py", line 30, in <module>
    import osmosdr
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\osmosdr\__init__.py", line 26, in <module>
    from osmosdr_swig import *
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\osmosdr\osmosdr_swig.py", line 17, in <module>
    _osmosdr_swig = swig_import_helper()
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\osmosdr\osmosdr_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_osmosdr_swig')
  File "C:\Program Files\GNURadio-3.7\gr-python27\lib\importlib\__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _osmosdr_swig

>>> Done

guruofquality, 2019-06-13:
Windows python is stupid, It will say no module if the DLLs arent in the path. Double check PATH or try running in the PothosSDR install's bin/ directory. Dependency walker can help as well.

keyboarderror, 2019-06-13:
Quite confusing. I wouldn't doubt there's a path conflict. That was the stand-alone install of GNURadio. The Pothos version will generate the GUI, connects to soapy it seems but still fails to send anything to the device.

Generating: 'C:\\SDR\\NBFM_Transmitter.py'

Executing: C:\Python27x64\python.exe -u C:\SDR\NBFM_Transmitter.py

gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.11.1

built-in sink types: uhd hackrf bladerf soapy redpitaya file 

[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.13.1.0-4-g56f966e2

-- Using subdev spec '0:0'.

gr::pagesize: no info; setting pagesize = 4096

[INFO] [UHDSoapyDevice] Using format CF32.

thread[thread-per-block[11]: <block gr uhd usrp sink (6)>]: UHDSoapyTxStream::send() = -5

guruofquality, 2019-06-13:
Since all of the drivers are installed, you have to pass device args to specify the device, its not using the hackrf driver in that snippet anyway. I think its hackrf=0 or something like that to tell gr-osmo to use the hackrf driver

keyboarderror, 2019-06-13:
specifying hackrf=0 doesn't change anything stand alone. Pothos version now fails with:

Generating: 'C:\\SDR\\NBFM_Transmitter.py'

Executing: C:\Python27x64\python.exe -u C:\SDR\NBFM_Transmitter.py

gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.11.1

built-in sink types: uhd hackrf bladerf soapy redpitaya file 

Traceback (most recent call last):

  File "C:\SDR\NBFM_Transmitter.py", line 282, in <module>

    main()

  File "C:\SDR\NBFM_Transmitter.py", line 270, in main

    tb = top_block_cls()

  File "C:\SDR\NBFM_Transmitter.py", line 152, in __init__

    self.osmosdr_sink_0 = osmosdr.sink( args="numchan=" + str(1) + " " + 'hackrf=0' )

  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\osmosdr\osmosdr_swig.py", line 1405, in make

    return _osmosdr_swig.sink_make(*args, **kwargs)

RuntimeError: Failed to open HackRF device (-5) HackRF not found

guruofquality, 2019-06-13:
Did you see this, does any of the software tools see the hardware, is this just a libusb setup issue (zadig)? https://github.com/pothosware/PothosSDR/wiki/Tutorial#recognizing-your-device

keyboarderror, 2019-06-14:
I've used Zadig many times. The find function shows it in all devices. Instantiating it shows the RTLs instead. Unplugging them causes it to appear but has no impact on GRC or anything. The older SDRAngel builds CAN access and make it work normally. That's now broken in newer versions and disabled. Not much else seems to work.

Logs:

C:\WINDOWS\system32>SoapySDRUtil.exe --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.13.1.0-4-g56f966e2
Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device

Found device 1
  default_input = False
  default_output = False
  device_id = 8
  driver = audio
  label = Digital Audio (S/PDIF) (High Definition Audio Device)

Found device 2
  default_input = False
  default_output = False
  device_id = 11
  driver = audio
  label = Microphone (High Definition Audio Device)

Found device 3
  default_input = False
  default_output = False
  device_id = 12
  driver = audio
  label = CABLE-B Output (VB-Audio Cable B)

Found device 4
  default_input = False
  default_output = False
  device_id = 13
  driver = audio
  label = CABLE-A Output (VB-Audio Cable A)

Found device 5
  default_input = False
  default_output = False
  device_id = 14
  driver = audio
  label = CABLE Output (VB-Audio Virtual Cable)

Found device 6
  default_input = False
  default_output = False
  device_id = 15
  driver = audio
  label = Headset Microphone (8- Wireless Controller)

Found device 7
  default_input = False
  default_output = False
  device_id = 16
  driver = audio
  label = VoiceMeeter VAIO3 Output (VB-Audio VoiceMeeter VAIO3)

Found device 8
  default_input = False
  default_output = False
  device_id = 17
  driver = audio
  label = VoiceMeeter Aux Output (VB-Audio VoiceMeeter AUX VAIO)

Found device 9
  default_input = False
  default_output = False
  device_id = 18
  driver = audio
  label = VoiceMeeter Output (VB-Audio VoiceMeeter VAIO)

Found device 10
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 325866e621553723
  part_id = a000cb3c005c4f6e
  serial = 0000000000000000325866e621553723
  version = git-509c8f4

Found device 11
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T

Found device 12
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 1
  serial = 00000001
  tuner = Rafael Micro R820T

*********************WITH RTLSDR*********************

C:\WINDOWS\system32>SoapySDRUtil.exe --make "driver=hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.13.1.0-4-g56f966e2
Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
  driver=Audio
  hardware=Audio
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

*********************WITHOUT RTLSDR*********************

C:\WINDOWS\system32>SoapySDRUtil.exe --make="driver=hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device driver=hackrf
[INFO] Opening HackRF One #0 325866e621553723...
  driver=HackRF
  hardware=HackRF One
  clock source=external
  part id=a000cb3c005c4f6e
  serial=0000000000000000325866e621553723
  version=git-509c8f4

guruofquality, 2019-06-14:
I wonder if you are seeing this issue, someone said that the pluto sdr driver was screwing up the one for rtl: https://github.com/pothosware/SoapyPlutoSDR/issues/24

As a quick test, try removing lib/SoapySDR/modules0.8/plutosupport.dll (its called something like that)

keyboarderror, 2019-06-14:
Yes. Removing PlutoSDRSupport.dll initializes the HackRF with RTL attached. GRC unchanged.

keyboarderror, 2019-06-19:
I've upgraded to PothosSDR-2019.06.09-vc14-x64. The device detection issue seems solved. Both GRC versions now respond HackRF not found.

guruofquality, 2020-09-01:
There is an open issue for this: https://github.com/pothosware/SoapyPlutoSDR/issues/24

And I brought it to the attention of the libIIO project: https://github.com/analogdevicesinc/libiio/issues/586

closed 2020-09-01 by guruofquality