pothosware/PothosSDR #74: Installation Interferes with GRC (Windows 7-64)

dbnut, 2020-01-23:
I _believe_ PothosSDR has stopped my GNURadio/GRC working. Is it designed to be stand-alone?

guruofquality, 2020-01-23:
Can you be more specific? It does set a few registry values for the python modules. And of course depending upon install options it does add to the environment variable %PATH%. If you don't change the PATH, the installer is standalone when launched from the shortcuts or running commands in the bin directory. What I usually see is other installs of Qt5 take precedence and mess up GQRX or PothosFlow.

 As with anything in the same search path -- one installers dlls can end up taking precedence over another's. And they might not be the same version of the dll or incompatible. So I would double check PATH and see what's in there.

dbnut, 2020-01-23:
Ah, thanks. Now I see that Pothos added its own GRC icon to the desktop, deleting mine.
Would be nice if the Pothos version was separate and distinguished?

guruofquality, 2020-12-28:
I dont know if I should rename it, but the desktop icons might be optional in the installer

closed 2020-12-28 by guruofquality