pothosware/PothosSDR #80: Add PothosNumPy to package

ncorgan, 2020-12-29:
https://github.com/pothosware/PothosNumPy

Notes:
- Requires PyYAML and Mako at build-time
- I'm not sure I've tested this with Windows before. I don't see why there would be a problem, but FYI

guruofquality, 2021-01-16:
@ncorgan little bit of template, type, and include compatibility fun

```

"C:\build\PothosSDR_vc16\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\build\PothosSDR_vc16\PothosNumpy.vcxproj" (default target) (46) ->
(CustomBuild target) ->
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(109,1): error C3646: 'getEpsilonExponent': unknown override specifier [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.
vcxproj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(109,35): error C2059: syntax error: '(' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(110,1): error C2334: unexpected token(s) preceding '{'; skipping apparent function body [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR
_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(119,1): error C3646: 'getMinExponent': unknown override specifier [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxp
roj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(119,31): error C2059: syntax error: '(' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(120,1): error C2334: unexpected token(s) preceding '{'; skipping apparent function body [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR
_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(134,1): error C3646: 'getNegativeEpsilonExponent': unknown override specifier [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\Poth
osNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(134,43): error C2059: syntax error: '(' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Cpp\NumericInfo.cpp(135,1): error C2334: unexpected token(s) preceding '{'; skipping apparent function body [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR
_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(155,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(165,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(177,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(191,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(96,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(107,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(249,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(260,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(271,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(284,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(299,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Build\PothosNumpy\BlockGen\BlockExecutionTestAuto.cpp(314,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
  C:\build\PothosSDR_vc16\Source\PothosNumpy\Testing\BlockExecutionTestManual.cpp(83,1): error C2059: syntax error: 'template' [C:\build\PothosSDR_vc16\Build\PothosNumpy\NumPyBlocks.vcxproj] [C:\build\PothosSDR_vc16\PothosNumpy.vcxproj]
[... cut ...]
eb761fbd59cbbc27f3b75e306f4824\PothosNumpy-configure.rule;C:\build\PothosSDR_vc16\CMakeFiles\73eb761fbd59cbbc27f3b75e306f4824\PothosNumpy-build.rule;C:\build\PothosSDR_vc16\CMakeFiles\73eb761fbd59cbbc27f3b75e306f4824\PothosNumpy-install
.rule;C:\build\PothosSDR_vc16\CMakeFiles\5e1b2df6da07058aa16011ab46323f30\PothosNumpy-complete.rule;C:\build\PothosSDR_vc16\CMakeFiles\e41d1ab7b2fec9b43f2c717d1ec9a580\PothosNumpy.rule' exited with code 1. [C:\build\PothosSDR_vc16\Potho
sNumpy.vcxproj]

    49 Warning(s)
    38 Error(s)
```

ncorgan, 2021-01-16:
Theoretically, this should be fixed on master, but I have issues building the Python stuff on my Windows partition.

guruofquality, 2021-01-16:
oh I was building master, didnt think about selecting the tag

ncorgan, 2021-01-16:
At the time, master was the tag, so same result. This is a commit added past that.