pothosware/PothosSDR #84: Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(C:\Program Files\PothosSDR\lib\Pothos\modules0.7\proxy\environment\PythonSupport.dll): failed safe load

otakenz, 2021-01-12:
OS: Windows 10 Education
Pothosware installer: Tried PothosSDR-2020.01.26-vc14-x64 and PothosSDR-2020.12.28-vc14-x64
Installation step: Add system variable to all user > C:\Program Files\PothosSDR >Full Installation
Pothosware usage: To develop custom python block for SDR Applications
Global python version: Python 3.9.1 (current) (numpy installed 1.19.5) (Other python versions I tried: Python 3.6.0, Python 3.6.8, Python 3.8.2..)
Swig-win version: swigwin-4.0.2

User Environment Path Variable:
[image]

System Environment Path Variable:
[image]

System Environment Variable:
[image]

Issue References:
https://github.com/pothosware/PothosSDR/issues/53
https://discourse.myriadrf.org/t/popthossdr-error-indicating-python-error/1636
https://groups.google.com/g/pothos-users/c/380EAoIj7AM/m/yKturFOmDQAJ
https://github.com/pothosware/PothosCore/issues/170

Doc References:
https://github.com/pothosware/PothosSDR/wiki/Tutorial
https://github.com/pothosware/SoapySDR/wiki/PythonSupport

Explanation:
From multiple sources, I heard that this PythonSupport.dll is a harmless warning however it is needed for python blocks development (https://github.com/pothosware/PothosSDR/issues/53), and that is the motive for me to try and solve the issue.
After days of trying, I am still unable to solve the issue, I have followed all the guidance I could find, but nothing helps. Can someone explain what this dll for, and does it really needed for python block development?

guruofquality, 2021-01-13:
Very thorough. I didnt realize python 3.9 is out, so the recent installer is probably expecting 3.8. I will update the build for the next installer.

So the trick is to install python3.8 for amd64 and to make sure python38.dll is found in the search path `%PATH%` (typically this is a python installer option). 

> Can someone explain what this dll for, and does it really needed for python block development?

PythonSupport.dll is essentially wrappers around Python.h library and API so Pothos runtime can call into python classes. Its linked against the python development library, hence why it needs the python dll in the path.

https://github.com/pothosware/PothosPython/blob/master/CMakeLists.txt#L96

guruofquality, 2021-01-30:
the last build should do python 3.9 now

closed 2021-01-30 by guruofquality