pothosware/PothosSDR #85: some questions about GRC

veso266, 2021-02-10:
hi there, it appears I missed a lot :)

not sure which gnuradio are you including in PothosSDR now (probably 3.8 am afraid to touch it since all my old xml based grc files will not work anymore (and I didn't find a converter for them yet) and not sure about what happens to old gnuradio blocks like gr-nrsc and gr-rds if they will still work :)

also noticied that python 3.9 is now required (will have to find a way to install it on my windows7 box (as someone up there decided that because windows 7 is out of support it cannot run python :( )

so yea, why python 3.9?

Sorry more like a question but issue :)

guruofquality, 2021-02-10:
There is a post FWIW It may explain some of this: https://groups.google.com/g/pothos-users/c/TVU_M7nQdxo

veso266, 2021-02-10:
Auch, quite a few changes

Guess I am staying on GRC 3.7 as GRC 3.9 has nothing I want even worse its missing stuff I want 😀

Radiomix2000, 2021-02-10:
This gnuradio update (3.7-3.8 etc...) was really piece of crap. Nobody care about compatibility. It was like somebody just wanted faster and faster send gr3.9 to production.. SAD.

veso266, 2021-02-10:
for GRC 3.6 to 3.7, at least I have a script that converts old flowgraphs
```sh
#!/bin/bash
# Version 0.2
# run ./grc_to_37 <appname>.grc
# this will create a backup copy and attempt to convert the api calls to 3.7
GRCFILE=$1
cp $GRCFILE $GRCFILE.pre3.7upgrade_backup # I refuse to take responsibility for breaking your flowgraphs
# set up a list of partial regexps to change modules
list_of_block_changes=(
"gr_peak_detector2_fb/blocks_peak_detector2_fb" 
"gr_short_to_float/blocks_short_to_float" 
"gr_diff_phasor_cc/digital_diff_phasor_cc" 
"blks2_logpwrfft_x/logpwrfft_x" 
"gr_rms_xx/blocks_rms_xx" 
"gr_probe_density_b/digital_probe_density_b" 
"gr_mute_xx/blocks_mute_xx" 
"gr_agc_xx/analog_agc_xx" 
"gr_ctcss_squelch_ff/analog_ctcss_squelch_ff" 
"gr_diff_encoder_bb/digital_diff_encoder_bb" 
"gr_pll_refout_cc/analog_pll_refout_cc" 
"gr_uchar_to_float/blocks_uchar_to_float" 
"gr_interp_fir_filter_xxx/interp_fir_filter_xxx" 
"gr_udp_sink/blocks_udp_sink" 
"gr_and_xx/blocks_and_xx" 
"gr_keep_one_in_n/blocks_keep_one_in_n" 
"gr_vector_sink_x/blocks_vector_sink_x" 
"gr_freq_xlating_fir_filter_xxx/freq_xlating_fir_filter_xxx" 
"gr_float_to_int/blocks_float_to_int" 
"gr_file_source/blocks_file_source" 
"gr_float_to_short/blocks_float_to_short" 
"gr_fft_vxx/fft_vxx" 
"gr_quadrature_demod_cf/analog_quadrature_demod_cf" 
"gr_file_sink/blocks_file_sink" 
"gr_pfb_clock_sync_xxx/digital_pfb_clock_sync_xxx" 
"gr_multiply_const_vxx/blocks_multiply_const_vxx" 
"gr_peak_detector_xb/blocks_peak_detector_xb" 
"gr_noise_source_x/analog_noise_source_x" 
"gr_simple_framer/digital_simple_framer" 
"gr_delay/blocks_delay" 
"gr_pfb_synthesizer_ccf/pfb_synthesizer_ccf" 
"gr_short_to_char/blocks_short_to_char" 
"gr_chunks_to_symbols_xx/digital_chunks_to_symbols_xx" 
"gr_interleaved_short_to_complex/blocks_interleaved_short_to_complex" 
"gr_frequency_modulator_fc/analog_frequency_modulator_fc" 
"gr_xor_xx/blocks_xor_xx" 
"gr_dpll_bb/analog_dpll_bb" 
"gr_threshold_ff/blocks_threshold_ff" 
"gr_nop/blocks_nop" 
"gr_unpack_k_bits_bb/blocks_unpack_k_bits_bb" 
"gr_max_xx/blocks_max_xx" 
"gr_int_to_float/blocks_int_to_float" 
"gr_stream_to_streams/blocks_stream_to_streams" 
"blks2_wfm_rcv/analog_wfm_rcv" 
"gr_fractional_interpolator_xx/fractional_interpolator_xx" 
"gr_conjugate_cc/blocks_conjugate_cc" 
"blks2_fm_demod_cf/analog_fm_demod_cf" 
"gr_fir_filter_xxx/fir_filter_xxx" 
"gr_null_source/blocks_null_source" 
"gr_complex_to_arg/blocks_complex_to_arg" 
"gr_pwr_squelch_xx/analog_pwr_squelch_xx" 
"gr_descrambler_bb/digital_descrambler_bb" 
"gr_copy/blocks_copy" 
"gr_complex_to_mag/blocks_complex_to_mag" 
"gr_throttle/blocks_throttle" 
"gr_additive_scrambler_bb/digital_additive_scrambler_bb" 
"gr_wavfile_sink/blocks_wavfile_sink" 
"gr_vector_source_x/blocks_vector_source_x" 
"gr_moving_average_xx/blocks_moving_average_xx" 
"gr_char_to_short/blocks_char_to_short" 
[... 64 lines cut ...]
"gr_vector_to_stream/blocks_vector_to_stream" 
"gr_agc2_xx/analog_agc2_xx" 
"gr_pll_carriertracking_cc/analog_pll_carriertracking_cc" 
"gr_repeat/blocks_repeat" 
"gr_filter_delay_fc/filter_delay_fc" 
"blks2_rational_resampler_xxx/rational_resampler_xxx" 
"gr_char_to_float/blocks_char_to_float" 
"gr_integrate_xx/blocks_integrate_xx" 
"gr_wavfile_source/blocks_wavfile_source" 
"gr_complex_to_float/blocks_complex_to_float" 
"gr_single_pole_iir_filter_xx/single_pole_iir_filter_xx" 
"blks2_nbfm_tx/analog_nbfm_tx" 
"blks2_wfm_rcv_pll/analog_wfm_rcv_pll" 
"const_source_x/analog_const_source_x" 
"blks2_pfb_arb_resampler_ccf/pfb_arb_resampler_xxx" 
)

for block in "${list_of_block_changes[@]}" 
do
    sed --in-place "/<key/s/$block/g" $GRCFILE
    # only replace the keys lines because otherwise
    # block ids can get clobbered if there are 2 of the same block
    # one old, and one new that both use grc generated ids
done
```

guruofquality, 2021-02-14:
Gnuradio has always been a disturbingly difficult part of this project to maintain. So many dependencies, moving targets, python modules...

The 3.9 release broke a ton of out of tree stuff and im waiting for much of it to be ported. It seems like they wanted to make some big changes to the project and im just trying to catch up to it. I even skipped really building 3.8 because of just being too busy.

And since I did this update build there are other kinks to work out too. People also wanted the upgrade too, so its a kind of damned if you do, damned if you dont situation. I hope things gradually shuffle into place.

So... I would consider the 2021 pothossdr builds as beta (but only as far as gnuradio support is concerned). Because 90% of the other stuff is pretty solid. I think even the gnuradio c++ part might be pretty solid too, since gqrx is using it. It seems that GRC and some out of tree projects are the main issue as everyone scrambles to "catch-up".

veso266, 2021-05-03:
Well after many consideratoins (list follows) I decided to upgrade to 3.9
Blocks that I use (almost all) work on gnuradio 3.9 or are ported to it
gr-nrsc5: https://github.com/argilo/gr-nrsc5
gr-zeromq (built-in now IIRC): https://github.com/gnuradio/gnuradio/tree/master/gr-zeromq
gr-drm: https://github.com/kit-cel/gr-drm (hmm is this built-in??)
gr-dvbt/t2: https://github.com/BogdanDIA/gr-dvbt (if reading this correctly, it built in now under gr-dtv)
gr-dvbs2: https://github.com/drmpeg/gr-dvbs2 (most build-in this works with 3.9)

Blcoks that don't work and I don't use that often (so can leave without)
gr-rftap https://github.com/rftap/gr-rftap (would probably try to port (and learn something along the way) this myself if only this would be missing :smile:)
gr-rds (don't know about this one but probably would work on 3.9): https://github.com/bastibl/gr-rds

gqrx should work with grnuradio 3.9
a lot of new flowgraphs and samples use yaml instead of xaml, and transpiling them back to xml by hand is.... (well its easier the other way around)

LEDs are tempting: https://wiki.gnuradio.org/index.php/QT_GUI_LED_Indicator
QT-GUI is similar to WX-GUI enough that I could probably adapt nicly

even found python3.9 that should work on windows 7: http://69.112.236.57/nextcloud/index.php/s/A6NGd76iMnMoHPm/download

so I booted my laptop with windows 8.1 up to test the new gnuradio first before upgrading and low and behold my upgrade path will be even harder

[image]

What kind of bluetooth api does gnuradio need? (why bluetooth, would thought some DSP things or idk audio apis or something would never suspect that bluetooth will be a problem :cry: (another problem I need to fix doh)

BTW: used latest pothos installer and python 3.9.4 (used the helper script cause I installed on fresh windows 8.1)