pothosware/PothosSDR #87: [Wishlist] Adding Codec2 to Gnuradio block tree

Radiomix2000, 2021-02-14:
From GR3.8 there is no codec2 pack included with Gnuradio. Is it possible to add the codec2 package into the last version of Pothos?

guruofquality, 2021-02-14:
Is this some codec2 development package, or did you have a particular gnuradio blockset in mind that uses it? Do you have a github or otherwise url for the project?

Radiomix2000, 2021-02-15:
Yes, for sure. 
This is official repo https://github.com/drowe67/codec2
And in the past the codec2 a was part of gr-vocoder package. But then they replaced this code beginning from GR3.8, and till now all GR versions (3.8.x, 3.9) are using system wide codec2 library (we speak about Linux). In Window everything works differently