pothosware/PothosSDR #90: Can't start GNURadio Companion

BMWE, 2021-04-30:
I've just installed PothosSDR-2021.02.28-vc16-x64 on my win 10 (ver 20H2).
Python installed as well
```
>python
Python 3.9.4 (tags/v3.9.4:1f2e308, Apr  4 2021, 13:27:16) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
```

When trying to start GNURadio Companion I'm getting following error.
Any suggestions to solve it?

[image]

guruofquality, 2021-04-30:
So one of the notes for python is that there is a checkbox (it might be in the advanced options menu) that will put an entry in the registry. `make sure that the "register extension" box is checked.` But I suppose it could have renamed since since I wrote that wiki. Its definitely an option on python3.9 installer though.
https://github.com/pothosware/PothosSDR/wiki/GNURadio#installation-instructions

I have been trying to make the executable search a few standard paths including local installs: https://github.com/pothosware/gnuradio-companion-exe/blob/master/gnuradio-companion.cpp#L68

But I may need to add to this logic. Just for my feedback, where is python.exe located for your system? Like `C:\Program Files\Python39\python.exe` for example

BMWE, 2021-04-30:
The python installation was made through Microsoft store.
I've tried to do the same with the python install I've downloaded, but
can't find such an option as you can see below. (I've checked python
versions 3.6 and 3.7 but couldn't see this option as well).

My python.exe is located either in C:\Python\Python39 or in C:\Program
Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1264.0_x64__qbz5n2kfra8p0
[image: image.png]
[image: image.png]

For windows, maybe it's a good idea to have python installation as part of
Pothos installation.
-----------------------------
Michael Poroger

"Science is not only knowledge, science is also to be dare"
Shimon Peres

guruofquality, 2021-04-30:
Thanks for the info

It looks like you need to check install for all users in the advanced or the HKEY_CURRENT_USER is written instead of HKEY_LOCAL_MACHINE: https://docs.microsoft.com/en-us/sql/machine-learning/install/media/python-install-for-all-users.png?view=sql-server-ver15

Action item for me: I need to change the executable to actually search both registry paths

If that checkbox doesnt do it, this is just a "helpful" launcher for GRC so you can manually run python.exe c:\Program Files\PothosSDR\bin\GNURadioHelper.py to get the dependencies installed and python.exe c:\Program Files\PothosSDR\bin\gnuradio_companion.py to run the application

BMWE, 2021-05-01:
Thanks for the tips. It worked.
I've extracted the key from HCU and imported the same to HLM and this was
solved.
-----------------------------
Michael Poroger

"Science is not only knowledge, science is also to be dare"
Shimon Peres