pothosware/PothosSDR #92: Pathos Flow - No Spectogram graph

Hagtronics, 2021-12-24:
Just a big thank you for putting this together. 

Downloaded PathosSDR installer (v0.8.0) for Windows 10 Pro (21H1) today, install is fine, GNU Radio Companion runs fine. Pathos Flow seems to be OK, no errors or warnings. When I use the simple example of a Square Wave Source and a Spectrogram, 'Activate' runs, and my CPU shows usage like something is running, but I don't get the 'Spectrogram Graph'. How to debug?

guruofquality, 2021-12-24:
https://github.com/pothosware/PothosFlow/wiki/Tutorial#insert-graphical-widget

Hagtronics, 2021-12-24:
Thanks - I'm an idiot! Can I delete my question? Ha, ha, ha...
I really appreciate the quick response and the work that went into making this package - it is the only working GNU Radio install That I was able to find for native Windows. While I could get other instances working I never could get my radio blocks to show up - with SoapySDR - all my radios are there and work.

closed 2021-12-24 by Hagtronics