pothosware/PothosSDR #94: trying to make OOT Block but missing doxygen and CPPUNIT

veso266, 2022-03-27:
Hello there, today I was trying to build an OOT module but figured out that pothos does not include (or I couldn't find it)
CPPUnit and DoxyGen

after installing Boost dev libraries from here: https://deac-ams.dl.sourceforge.net/project/boost/boost-binaries/1.67.0/boost_1_67_0-msvc-14.0-64.exe

added theese variables
```
BOOST_INCLUDEDIR    C:\Program Files\PothosSDR\boost
BOOST_LIBRARYDIR    C:\Program Files\PothosSDR\boost\lib64-msvc-14.0
BOOST_ROOT          C:\Program Files\PothosSDR\boost\boost
```
(nothing worked exept theese, I tried first like this: cmake ../ -DBOOST_ROOT="C:\Program Files\PothosSDR\boost\boost"  -DBOOST_LIBRARYDIR="C:\Program Files\PothosSDR\boost\lib64-msvc-14.0"

but it didn't work, only setting them as variables worked

so now I am only missing CPPUNIT and doxygen, I hope someone has them precompiled (I couldn't find them in pothos install dir like most of the other libraries) because I am trying to avoid building them from source if absolutely not necessary

Thanks for Anwsering and Best Regards

guruofquality, 2022-03-27:
You might have better luck copying how the build does it: https://github.com/pothosware/PothosSDR/blob/master/BuildGNURadio.cmake#L76 Because it is kind of just arbitrary magic:

## BOOST

So every cmake project gets the following boost flags:
```
        -DBOOST_ROOT=${BOOST_ROOT}
        -DBOOST_LIBRARYDIR=${BOOST_LIBRARYDIR}
        -DBOOST_ALL_DYN_LINK=TRUE
```
     
And how those variables are set
* https://github.com/pothosware/PothosSDR/blob/master/InstallPrebuilt.cmake#L34

## CPPUnit

```

        -DCPPUNIT_INCLUDE_DIRS=${CPPUNIT_INCLUDE_DIRS}
        -DCPPUNIT_LIBRARIES=${CPPUNIT_LIBRARIES}
```

And cpp unit is definitely installed for development (unlike boost)
So CPPUNIT_INCLUDE_DIRS should be C:\Program Files\PothosSDR\include\cppunit
and CPPUNIT_LIBRARIES should be C:\Program Files\PothosSDR\lib\cppunit_dll.lib

veso266, 2022-03-28:
Yey, thanks that did the trick now this thing finaly configures (pain follows :cry:)

Let me tell you a story :smile:
I was missing this: https://github.com/gnuradio/gnuradio/tree/master/gr-video-sdl
I guess libsdl wasn't present during buildtime so it was skipped

So I decided to add this myself, how hard could it be.....
Well since I am still using [GNURadio 3.7.14](https://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2020.12.28-vc14-x64.exe) (because I eather get one module working and be done, or upgrade to 3.9 and face a lot more trouble along the way) I pulled the release and pulled gr-video-sdl out of it, so I decided to stuff it into its own OOT module and call it a day

well my instruction speak for itself :smile:
https://github.com/veso266/gr-video_sdl/blob/main/Instructions.MD

now the first part (configure part) is done
now the building follows......

veso266, 2022-03-28:
Yey, thanks that did the trick now this thing finaly configures (pain follows :cry:)

Let me tell you a story :smile:
I was missing this: https://github.com/gnuradio/gnuradio/tree/master/gr-video-sdl
I guess libsdl wasn't present during buildtime so it was skipped

So I decided to add this myself, how hard could it be.....
Well since I am still using [GNURadio 3.7.14](https://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2020.12.28-vc14-x64.exe) (because I eather get one module working and be done, or upgrade to 3.9 and face a lot more trouble along the way) I pulled the release and pulled gr-video-sdl out of it, so I decided to stuff it into its own OOT module and call it a day

well my instruction speak for itself :smile:
https://github.com/veso266/gr-video_sdl/blob/main/Instructions.MD

now the first part (configure part) is done
now the building follows......

Not sure how you are building this but since configure didn't generate any MakeFile I couldn't use make so I opened generated solution in Visual Studio 2019 and if only it would give me some normal errors :cry:
[image]
it seams that everything but my code is complaining :cry: (configure didn't complain, why does linker need to complain now....)

but it did build 8 of them fine
[image]

still haven't find any files to copy over to gnuradio dir

veso266, 2022-03-29:
I managed to solve linker errors for SDL (copying it into PothosSDR/include/SDL and PothosSDR/lib did the trick) I guess theese folders were in path or something

but fun didn't last for long :cry: 
[image]
I guess I have some library mismatch or something

If I am going through so much trouble I might as well upgrade to 3.9 along the way and add stuff I am missing there
(I managed to install it on Windows 7 but most of the stuff (I use) are missing as well as [gr-video-sdl](https://github.com/gnuradio/gnuradio/tree/master/gr-video-sdl)

so lets try to ugrapde to 3.9 and add gr-rds and gr-nrsc5 to it and see where we come :smile:

veso266, 2022-04-02:
So close, yet so far away, I've got everything built exept SoapySDR complains for something (its looking python in its own source directory for some reason ....) (not sure what to put in that directory because I already have python3.9 in C:\Python3.9)

and all the variables
```
-DPYTHON_EXECUTABLE=${Python_EXECUTABLE}
        -DPYTHON_INCLUDE_DIR=${Python_INCLUDE_DIRS}
        -DPYTHON_LIBRARY=${Python_LIBRARIES}
        -DPYTHON_INSTALL_DIR=${PYTHON3_INSTALL_DIR}
``` 
that he expects points to python (exept PYTHON3_INSTALL_DIR, which is empty... not sure where its supposed to point to?)

[image]
 
as for GNURadio its still doesn't start, it complains that it cannot import GRPython but all the paths point to it just like if using your installer

veso266, 2022-04-02:
Aha, I think I found the module not importing error in GNURadio
[image]

veso266, 2022-04-03:
Hi there, after trying to freshly build this again (because I thought I destroyed something along the way) and waiting 19 hours again (yea, I am using VM for this so that's why its so slow...) I managed to spot and fix GR-SDRPlay along the way (because script was looking for master branch which is now main (I told it to clone [maint-3.9](https://github.com/fventuri/gr-sdrplay3/tree/maint-3.9) (just to be on the safe side)) 
now have the following problems:

GROsmoSDR:
[image]
Its looking for gnuradio.bindtoolcuz its trying to import BindingGenerator but I checked and I cannot find this anywhere
it says [here](https://osmocom.org/issues/5021) that GROsmoSDR should be supported on GNURadio so I have no idea why it doesn't want to build

PothosPlotters: 
[image]
What does this do? (can I turn it off somehow, I only want GNURadio :smile:)

SoapySDRPython3:
[image]
Its looking for python directory in its sources which is nowhere to be found (even on github its not there)

I did find its included from here: https://github.com/pothosware/SoapySDR/blob/9cbaa3cfb4ad6231a743319d71bda3634885a428/swig/python3/CMakeLists.txt#L65

but removing that include causes even more pain
[image]

as for GNURadio its still doesn't run :cry:
[image]
IMPORT_GR still failes, because ``gr_python.cp39-win_amd64.pyd`` is still looking for this mpir.dll which I don't know why it haven't got installed (I saw it built)

So if you maybe have any idea what is wrong (at least Pothos stuff, GROsmoSDR is probably broken somehow)?

Thanks