pothosware/PothosSDR #95: PothosSDR does not fully uninstall from windows 10

lifeatthesharpend, 2022-03-28:
In windows 10 when I tried to uninstall the app never fully uninstalls and remains in the "Apps & Features" dialog box of windows 10. The SW is "Pothos SDR environment (vc16-x64)" on windows 10. After uninstall this still shows in the settings -> Apps & features
[image]

once selecting uninstall the dialog box shows : Windows cannot find'C:\Projeram files\PothosSDR\Uninstall.exe' Make sure you typed the name correctly, and then try again.

[image]

The folder does not exist. but the OS still thinks it is installed.

colaflaske, 2022-04-02:
The cause of this is an orphaned registry entry at 
`HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\PothosSDR`.

If the folder C:\Program Files\PothosSDR is successfully deleted after uninstalling, then deleting the PothosSDR folder in the registry using regedit should be no problem. (disclaimer: don't normally mess with your registry)

lifeatthesharpend, 2022-04-02:
that registry entry does not exsit
[image]

and Pothos SDR environment is still there:
[image]
The folder C:\Program Files\PothosSDR is not present

lifeatthesharpend, 2022-04-02:
After doing a search for PothosSDR in the registry it found this:
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Microsoft\Windows\CurrentVersion\Uninstall\PothosSDR
[image]

after deleting the reg entry and substrings PothosSDR does look to be removed from the windows "Apps & features" dialog box